opencog/ros-behavior-scripting #14: chat problem fixed (need to publish chatbot_affect_express) but can't push

demaris, 2016-01-20:
Chatbot was in a wait state for the chatbot_affect_express when -sent flag used for sentiment detection. 
Responses wouldn't go to tts when polarity was detected in this case. 
I don't seem to have permission to be a committer,  sent fix by email.

linas, 2016-02-18:
Hi David, I missed this. Was your fix committed?  Is the issue resolved, or is it still outstanding?

You should have permission to create a pull request; I can merge these.

demaris, 2016-02-18:
Hello from beautiful Louisville, where I'm mud wrestling Phil on the banks
of the OHIGHO.
I'm not sure if I quite remember the situation, but I just tried again. I
can't push to master or to a branch.
ERROR: Permission to opencog/ros-behavior-scripting.git denied to demaris.
So I think it's still outstanding and anyone using chat with -sent will get
responses only if polarity is neutral (polarity score below threshold).

linas, 2016-02-18:
Of course you can't push to master or to branch. you are not supposed to!  Push to your own repo, and create a pull request.

linas, 2016-03-03:
Closing .. I have not seen a pull request.

closed 2016-03-03 by linas