opencog/ros-behavior-scripting #18: btree.scm can't find eva-behavior?

demaris, 2016-03-02:
[image]

demaris, 2016-03-02:
Just did a fresh install, no build errors but hit this.  I do see some things that look like paths in eva-behavior.scm while the files are all in the same directory, but I don't know the semantics of the load.

wenwei-dev, 2016-03-02:
Hi Dave,

I haven't tried the latest build. What we do in HR is using specific
revisions of OpenCog stack. Using the latest code may not work.

I've pushed a new script called "hrtool" to scripts repo. The aim is to
substitute the existing code management scripts.

I've tested on a VM, and it worked. So I think you may want to try it.

By running ./hrtool -G will get all the HR and OC source code and set them
to the right revisions.

wenwei-dev, 2016-03-02:
Hi again,

It seems you need to build and install ros-behavior-scripting using cmake.
Linas mentioned it in an earlier email.

So -- you need to either update Docker files, shell scripts, etc. to add a
> 'mkdir build; cd build; cmake ..; make; make install' in
> the ros-behavior-scripting directory, or .. simply don't pull the latest
> code.

demaris, 2016-03-02:
So instead of setup.sh --install_dev I would use that?

If I want to do a pull later I just do the individual repos and then
setup.sh?  Or use .hrtool without flags?

demaris, 2016-03-02:
I guess I assumed that that was now part of our setup.sh,  at least if the
--install_dev.

I'll do it.  I saw it in the .md file but I assumed this was intended for
Linas's standalone environment for unit test or using the old public repos.

demaris, 2016-03-02:
If we are trying to use a particular build, could we just fork a build into
our public_ws,  and have ./hrtool -G install
that instead of getting the latest from openCog?
To me it would make more sense that this is an application of openCog and
would just live in our space.

wenwei-dev, 2016-03-02:
Hi Dave,

The cmake steps for OC btree are not added to any of the scripts yet.
Sorry, I'll added it soon.

hrtool should be able to pull the latest code for HR. It skips updating OC
just because it's not stable sometimes. But now since it's getting better,
I'll make it to update OC source code as well.

I'm also thinking that we fork OC to our repo to make it clearer.

Thanks,
Wenwei

linas, 2016-03-03:
Ah hah, so this is what you were talking about. 

-- I did send out a warning that this repo is turbulent, and will continue to be into the indefinite future.

-- this is where tagging the repo with a tag, e.g. "ok-for-production-use" tag, would be ideal.  You would only pull from that tag, and then, every few weeks, when you have time to test and verify, you could move that tag forwards to get the newer code.

linas, 2016-03-03:
am closing, as I assume that everything is back to a working state, now.

BTW, *today's* version of this code also requires *today's* version of atomspace and opencog.   This is not usually the case: these three repos can often be used "out of sync". but sometimes there are coordinated changes going into all three.

closed 2016-03-03 by linas