opencog/ros-behavior-scripting #35: Btree can't handle message from topic "chatbot_affect_express"

wenwei-dev, 2016-03-15:
[rosout][ERROR] [1458036392.794756]: bad callback: <bound method EvaControl.chatbot_affect_perceive_cb of <ros_commo.EvaControl instance at 0x7f3fd32acea8>>
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/topics.py", line 720, in _invoke_callback
    cb(msg)
  File "ros_commo.py", line 243, in chatbot_affect_perceive_cb
    self.puta.chatbot_affect_negative()
  File "put_atoms.py", line 68, in chatbot_affect_negative
    scheme_eval_h(self.atomspace, "(State chat-affect chat-negative)")
  File "scheme_wrapper.pyx", line 47, in opencog.scheme_wrapper.scheme_eval_h (/home/wenwei/workspace/hansonrobotics/opencog/atomspace/build/opencog/cython/opencog/scheme_wrapper.cpp:1148)
RuntimeError: Backtrace:
In ice-9/boot-9.scm:
 157: 12 [catch #t #<catch-closure 25541e0> ...]
In unknown file:
   ?: 11 [apply-smob/1 #<catch-closure 25541e0>]
In ice-9/boot-9.scm:
 157: 10 [catch #t #<catch-closure 25540c0> ...]
In unknown file:
   ?: 9 [apply-smob/1 #<catch-closure 25540c0>]
   ?: 8 [call-with-input-string "(State chat-affect chat-negative)" ...]
In ice-9/boot-9.scm:
2320: 7 [save-module-excursion #<procedure 23b0cf0 at ice-9/eval-string.scm:65:9 ()>]
In ice-9/eval-string.scm:
  44: 6 [read-and-eval #<input: string 2557680> #:lang ...]
  37: 5 [lp (State chat-affect chat-negative)]
In ice-9/eval.scm:
 387: 4 [eval # ()]
 393: 3 [eval #<memoized chat-affect> ()]
In unknown file:
   ?: 2 [memoize-variable-access! #<memoized chat-affect> #<directory # 1088c60>]
In ice-9/boot-9.scm:
 102: 1 [#<procedure 26a5800 at ice-9/boot-9.scm:97:6 (thrown-k . args)> unbound-variable ...]
In unknown file:
   ?: 0 [apply-smob/1 #<catch-closure 2554080> unbound-variable ...]

ERROR: In procedure apply-smob/1:
ERROR: Unbound variable: chat-affect
ABORT: unbound-variable
 (/home/wenwei/workspace/hansonrobotics/opencog/atomspace/opencog/guile/SchemeEval.cc:885)

linas, 2016-03-16:
Pull request #36 should fix this.

wenwei-dev, 2016-03-22:
I patched this fix to HR btree, but the same error still raised.

linas, 2016-07-12:
Is this still an issue, or can this be closed?

wenwei-dev, 2016-07-15:
Yes, you can close it. I don't see the same issue. But there is another issue about handling speech event #69

closed 2016-07-17 by linas