opencog/ros-behavior-scripting #59: netcat times out

yantrabuddhi, 2016-07-11:
I made pull request for octomap integration, however the netcat function times out and breaks face follow function. I did it another hacky way using the same scheme function in a thread in ros_commo and that way the robot functions okay.

yantrabuddhi, 2016-07-11:
My observation is that after running ./hrtool -B and then launching the ./dev.sh makes the robot work okay, but when the program is stopped and started, then on the second try without running ./hrtool -B, the program fails on netcat

linas, 2016-07-12:
This package does not contain any scripts called `dev.sh` or `hrtool` -- you should open the bug report against the component that actually has the bug in it!

I am guessing its a bug in the hrtool.   I am guessing that when you stop and start, you end up creating a cogserver listening at a different port, instead of using the configured port; as a result netcat cannot talk to the cogserver, and fails. 

This is how TCPIP is designed to work, on purpose: otherwise, left-over tcpip packets from a previous connection, that are still floating around on the internet somewhere, arrive at the new server and garble up the socket. 

http://blog.davidvassallo.me/2010/07/13/time_wait-and-port-reuse/
http://arstechnica.com/civis/viewtopic.php?t=1130755
https://vincent.bernat.im/en/blog/2014-tcp-time-wait-state-linux.html

Spending some time with `tcpdump` will clarify what the issue is.

closed 2016-07-12 by linas

bgoertzel, 2016-07-13:
Hmmm... Linas, you may well be right about the source of the bug.  And I
guess you are right that Mandeep did not open the issue in the right place.

However, right now this practical problem is very serious and severe, and
is blocking our progress toward giving an OpenCog/Hanson Robotics demo, and
Mandeep has spent a lot of time trying to track down the bug without any
success

So if you or Wenwei or anyone can help solve this problem very soon that
will be much appreciated

Otherwise we will have to do some radical workaround, like just replace a
bunch of Scheme and python code with C++ that is easier for Mandeep to
debug -- but then this will take a lot of time too and may result in us not
getting our demo done...

bgoertzel, 2016-07-13:
Haha, looks like Mandeep fixed the bug, it was a simple typo 8-D

bgoertzel, 2016-07-13:
3 days of hacking spent .. more or less ... trying to work around a bug
caused by \ instead of \\   ...

yantrabuddhi, 2016-07-13:
Hi Linas,
 i found the issue. I did not expect it though. Netcat requires two
trailing \n\n , i was giving just one \n, I just ccould not think this
could be a problem.

linas, 2016-07-13:
netcat does not care at all about carriage returns or newlines, they are just like any other bytes, as far as netcat is concerned.

It is possible that the cogserver socket is eating up random carriage returns.  The cogserver socket interfaces are wildly over-engineered, excessively complicated, and difficult to use. They do crazy things like add and remove newline characters, and behave differently depending on whether there is a newline or not.  Certainly, extra newlines will not hurt, in this case.

However, that code is stable, no one touches it ever, and its been working great for 5+ years, so it is very unlikely that is the cause of the problem.  

Therefore, I would look at the message being sent, and try to understand how it is different from the other messages.  The message is calling into the brand-new octotree code .. perhaps its the octreee that is hanging, and because it hangs, it never returns to scheme, which never retuirns to the cogserver, which never closes the socket.

So, please verify that the octotree is actually getting the message, and that it is not hanging.

If you are creating threads with a socket still open, please make sure that your other threads aren't doing weird things, like sending data to scheme, which is then trying to buffer it to a socket that is being managed in a different thread.

Random prints in your octree code could possibly mangle the socket, since these get buffered up to be returned to netcat.

yantrabuddhi, 2016-07-13:
Hi Linas,
 I verified that existing netcat strings had two carriage returns, and only
thing different in my string was single carriage return. When I chaged from
single carriage return to two consecutives ones at end the code works
properly without issues.