opencog/ros-behavior-scripting #68: netcat still times out in face_atomic.py

yantrabuddhi, 2016-07-14:
In face track if I update cogserver at 30 hz or 15 hz, netcat starts blocking and times out. I tries doing this by also just sending "(ConceptNode "12345678")" at 15 hz and it still causes netcat to time out and btree to stop

yantrabuddhi, 2016-07-14:
Test code to crash guile cogserver after stopping netcats from face_atomic.py
    import time
    from netcat import netcat

    while (True):
        netcat("localhost",17020,"(ConceptNode (\"1234567\")\n")
        print "sent...\n"
        time.sleep(0.03)

linas, 2016-07-14:
OK, I can reproduce this.

linas, 2016-07-16:
pull requests #2288 #2289 #2290  #2291 resolve some of this issue.

There seems to be a memleak of some sort, as the cogserver keeps growing, and keeps getting slower. Trying to track that down.

linas, 2016-07-16:
I was wrong, there is non memleak -- valgrind confirms this. What was happening is that guile was slowly growing its heap; after a certain point, it settles down.

Note that the example program above has unbalanced parenthesis; this generates guile errors at a rate of 30/second. 

When the parenthesis are balanced, the system seems to mostly work most of the time, but I can still sometimes hang it -- specifically, in `boost::asio::recv` -- it hangs, waiting for data; I cannot tell why.  Investigating.

linas, 2016-07-17:
I think I eliminated the last of the crashes with pull request opencog/opencog#2296 

Please pull, recompile, and test, and close this bug if it starts working for you.

linas, 2016-07-18:
Mandeep reports that this is now working for him, am closing bug.

closed 2016-07-18 by linas