opencog/ros-behavior-scripting #69: Unbound variable "chat-state"

wenwei-dev, 2016-07-15:
Run this to reproduce the error
`rostopic pub -1 /robot/speech_events std_msgs/String "start"`

```
ERROR: In procedure apply-smob/1:
ERROR: Unbound variable: chat-state
ABORT: unbound-variable
```

linas, 2016-07-17:
The module `eva-mdeol` defines this.   Some subsystem failed to  `(use-modules (opencog eva-model))`  Not sure which.

linas, 2016-07-17:
Anyway .. is this still an issue? There have been a log of changes to the chatbot in the last 3 days; this might be fixed already?

wenwei-dev, 2016-07-18:
I just test it, and the bug is still there.

leungmanhin, 2016-07-19:
Strange, I didn't get this error, was the ros-behavior-scripting running?

wenwei-dev, 2016-07-19:
Yes, it was running. It happened when I chatted with it. But this is not
urgent and important.

linas, 2016-09-21:
Closing, I assume this is no longer an issue

closed 2016-09-21 by linas