opencog/ros-behavior-scripting #74: netcat to cogserver in btree is failing too often

yantrabuddhi, 2016-07-21:
I updated and repeated with @amen  and found netcat blocking and cogserver crash happening regularly. @linas told a way to see things with gdb to @misgana . We observed that when we run btree-psi from gdb at the end after launching all other nodes the problem disappears (face_track.py needs to be killed and restarted once for some reasons). However gdb runs the stuff very slowly and gets slower over time (maybe due to loading lots of atoms?) but nothing crashes for as long as I observed [60 minutes or so]

All the above happens and problems are observed by running ./dev.sh --oc --cmt in HEAD repository scripts folder, running normally with updated ros-behavior-scripting master branch.
I have so far not found the changes I made over last working state to cause this. So far my guess is some problem with cogserver with netcat. Also running guile -l btree-psi.scm at end without gdb at times make the system work without problem. if we add clear face trackers ros message for face tracker and run btree at end it can solve the problem

yantrabuddhi, 2016-07-21:
Seems like I can't replicate it on other machines, but cogserver is getting blocked for me and others for reasons we can't pin point

closed 2016-07-21 by yantrabuddhi

AmeBel, 2016-07-21:
Using the latest opencog, rbs, HEAD and using `.gdbinit` from [here](https://github.com/opencog/docker/blob/master/opencog/base/scripts/.gdbinit), the folowoing are the steps.

1. Modified https://github.com/hansonrobotics/HEAD/blob/master/scripts/dev.sh#L107 to 
`  tmux new-window -n 'cog' "export ROS_NAMESPACE=$NAME; cd $OCBHAVE/src; gdb guile -l btree-psi.scm; $SHELL"`
2. run `./dev.sh --oc --cmt` from https://github.com/hansonrobotics/HEAD/tree/master/scripts directory
3. once the tmux session opens, change to `cog` window.
4. In the gdb shell , run ` run `
5. In the guile shell that becomes available run `(load "btree-psi.scm")`
6. In another terminal, telnet to the cogserver then enter the scm schell and run `(count-all)` __It should
hang__
7. interrupt the process running in gdb by `Ctrl + c` 
(__Next steps follows suggestion from https://github.com/opencog/opencog/issues/2301#issuecomment-233740712__)
8. `info threads` gives
[image]

9. output from running `thread n` and `bt` for n = { 288, 287, 286, 285, 284} 
[image]
[image]