opencog/ros-behavior-scripting #82: cog-delete always consistently found in random btree crash

yantrabuddhi, 2016-08-17:
Hi Linas,
 I checked the string it was (cog-delete (NumberNode "16405")). We tried several runs and problem was always with cog-delete. We removed cog-delete netcats from face-atomic.py and the system did not crash for 20 minutes of total test time.
Below is stack trace when cog-delete was being sent
stack trace:
#0  opencog::AtomTable::getHandle (this=this@entry=0x0, t=t@entry=8, 
    name=<incomplete type>) at /usr/include/c++/4.8/bits/stl_vector.h:791
#1  0x00007feaa44d6a24 in opencog::AtomSpace::add_node (this=0x0, t=t@entry=8, 
    name=..., async=async@entry=false)
    at /home/hr/hansonrobotics/opencog/atomspace/opencog/atomspace/AtomSpace.cc:313
#2  0x00007feaa471949d in opencog::SchemeSmob::ss_new_node (
    stype=<optimized out>, sname=0x1d3c9c0, kv_pairs=0x304)
    at /home/hr/hansonrobotics/opencog/atomspace/opencog/guile/SchemeSmobNew.cc:370
#3  0x00007feaa6c19111 in ?? () from /usr/lib/libguile-2.0.so.22
#4  0x00007feaa6b9b747 in scm_call_1 () from /usr/lib/libguile-2.0.so.22
#5  0x00007feaa6c1912a in ?? () from /usr/lib/libguile-2.0.so.22
#6  0x00007feaa6b9b81e in scm_call_3 () from /usr/lib/libguile-2.0.so.22
#7  0x00007feaa6c1912a in ?? () from /usr/lib/libguile-2.0.so.22
#8  0x00007feaa6b9b863 in scm_call_4 () from /usr/lib/libguile-2.0.so.22
#9  0x00007feaa471ae71 in opencog::SchemeEval::do_eval (this=0x7fe968061540, 
    expr=...)
    at /home/hr/hansonrobotics/opencog/atomspace/opencog/guile/SchemeEval.cc:655
#10 0x00007feaa471af93 in opencog::SchemeEval::c_wrap_eval (p=0x7fe968061540)
    at /home/hr/hansonrobotics/opencog/atomspace/opencog/guile/SchemeEval.cc:583
---Type <return> to continue, or q <return> to quit---q
Quit
(gdb) bt
#0  opencog::AtomTable::getHandle (this=this@entry=0x0, t=t@entry=8, 
    name=<incomplete type>) at /usr/include/c++/4.8/bits/stl_vector.h:791
#1  0x00007feaa44d6a24 in opencog::AtomSpace::add_node (this=0x0, t=t@entry=8, 
    name=..., async=async@entry=false)
    at /home/hr/hansonrobotics/opencog/atomspace/opencog/atomspace/AtomSpace.cc:313
#2  0x00007feaa471949d in opencog::SchemeSmob::ss_new_node (
    stype=<optimized out>, sname=0x1d3c9c0, kv_pairs=0x304)
    at /home/hr/hansonrobotics/opencog/atomspace/opencog/guile/SchemeSmobNew.cc:370
#3  0x00007feaa6c19111 in ?? () from /usr/lib/libguile-2.0.so.22
#4  0x00007feaa6b9b747 in scm_call_1 () from /usr/lib/libguile-2.0.so.22
#5  0x00007feaa6c1912a in ?? () from /usr/lib/libguile-2.0.so.22
#6  0x00007feaa6b9b81e in scm_call_3 () from /usr/lib/libguile-2.0.so.22
#7  0x00007feaa6c1912a in ?? () from /usr/lib/libguile-2.0.so.22
#8  0x00007feaa6b9b863 in scm_call_4 () from /usr/lib/libguile-2.0.so.22
#9  0x00007feaa471ae71 in opencog::SchemeEval::do_eval (this=0x7fe968061540, 
    expr=...)
    at /home/hr/hansonrobotics/opencog/atomspace/opencog/guile/SchemeEval.cc:655
#10 0x00007feaa471af93 in opencog::SchemeEval::c_wrap_eval (p=0x7fe968061540)
    at /home/hr/hansonrobotics/opencog/atomspace/opencog/guile/SchemeEval.cc:583
#11 0x00007feaa6b922ca in ?? () from /usr/lib/libguile-2.0.so.22
#12 0x00007feaa6c1912a in ?? () from /usr/lib/libguile-2.0.so.22
#13 0x00007feaa6b9b863 in scm_call_4 () from /usr/lib/libguile-2.0.so.22
#14 0x00007feaa6b929ff in ?? () from /usr/lib/libguile-2.0.so.22
#15 0x00007feaa6b92a95 in scm_c_with_continuation_barrier ()
   from /usr/lib/libguile-2.0.so.22
#16 0x00007feaa6c0b78c in ?? () from /usr/lib/libguile-2.0.so.22
#17 0x00007feaa6300fe2 in GC_call_with_stack_base ()
   from /usr/lib/x86_64-linux-gnu/libgc.so.1
#18 0x00007feaa6c0bb48 in scm_with_guile () from /usr/lib/libguile-2.0.so.22
#19 0x00007feaa471af3e in opencog::SchemeEval::eval_expr (this=0x7fe968061540, 
    expr=...)
    at /home/hr/hansonrobotics/opencog/atomspace/opencog/guile/SchemeEval.cc:551
#20 0x00007feaa3991a60 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#21 0x00007feaa6921184 in start_thread (arg=0x7fe97e7fc700)
    at pthread_create.c:312
#22 0x00007feaa664e37d in clone ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111

sta

linas, 2016-08-17:
OK, since this is almost surely not a bug ros-behavior-scripting, it would be better to open the bug report against the component that is crashing -- in this case, the atomspace, or the cogserver.  But anyway, for right now, lets leave this bug report here, for now.

linas, 2016-08-17:
Next, the stack trace clearly indicates that the atomspace pointer is null.  Now, the atomspace pointer is normally obtained from thread-local storage, in the C function ss_get_env_as() at line 348, 349.

If the thread-local storage is not initialized, this suggests that the thread itself was never initialized.  I guess that the cog-delete is being sent from somewhere in here, in ros-behavior-scripting?

yantrabuddhi, 2016-08-17:
Hi Linas,
 I am back at home now, so I can't check further. However*, *whenever there
was crash the string expression had cog-delete in it. I wasn't sure where
to check if thread wasn't properly initialized. I was reading code in
GenericShell.cc but I couldn't get to the root cause

linas, 2016-08-17:
I assume its line 98 of face_atomic.py is the location of the cog-delete.

yantrabuddhi, 2016-08-17:
yes, also when i commented line 98 then it crashed again and the cog-delete
string before line 98 was shown at the time in stack trace from core dump.
cog-delete (list (NumberNode "12345"))

linas, 2016-08-17:
I'm making the wild guess that pull request opencog/opencog#2377 might fix the problem. Not quite sure, though.

linas, 2016-08-17:
By the way, `(cog-delete (list .... anything` is bad syntax. cog-delete expects an atom, not a list.

yantrabuddhi, 2016-08-17:
sorry I meant ListLink

yantrabuddhi, 2016-08-17:
this is the exact thing that we commented the netcat out to test
def delete_face(self, faceid):
pattern = "(EvaluationLink (Predicate \"name\") (ListLink (ConceptNode \""
+ str(faceid) + "\") (VariableNode \"f\")))"
del_face = "(cog-execute! (PutLink (DeleteLink "+pattern+") (GetLink
"+pattern+")))"
face = "(cog-delete " + \
"(EvaluationLink (PredicateNode \"visible face\") " + \
"(ListLink (NumberNode \"" + str(faceid) + "\"))))\n" + \
del_face + "\n" + \
"(cog-delete " + \
"(ListLink (NumberNode \"" + str(faceid) + "\")))\n" + \
"(cog-delete (NumberNode \"" + str(faceid) + "\"))\n"
return face

yantrabuddhi, 2016-08-18:
The crash problem seems to be solved with the change opencog/opencog#2377

closed 2016-08-18 by yantrabuddhi