opencog/ros-behavior-scripting #83: tracking stops in case of single detected face after face being lost and re-detected

yantrabuddhi, 2016-08-18:
When we run the behavior tree from hanson HEAD repository, initially the robot detects the face and tracks it. However on loosing the track of face and re-detecting any single face the tracking does not happen. When there are more faces detected however the tracking starts working while blender head moves in strange way. 
An observation is that this happens when face is re-detected and exactly after the glance at new face happens a someone left rule is triggered and then the tracking does not happen

linas, 2016-08-18:
@AmeBel or @rtrautlein shoul dbe able to help with that. The psi rules should be identical to the old behavior tree riles, they were not changed.  So an obvious thing to ch3eck to to make sure that the visible faces state is changing correctly when faces appear and disappear.

Commonting out the delete-atom clauses will definitely mess up face tracking.

linas, 2016-09-21:
Is this still an issue?

AmeBel, 2016-09-22:
issue is resolved.

closed 2016-09-22 by AmeBel