opencog/ros-behavior-scripting #92: Behavior tree (and actually the whole Guile) crashes with a Floating point exception

leungmanhin, 2016-09-05:
This keeps happening today when we try to run it for a demo, everything are the latest on master branches, and we do all the code updating, cleaning, building and installing by using the `hrtool` in `private_ws/scripts`

## **To reproduce the error**

- Start the stack, with a webcam plugged in, using `robot.sh` in the same directory
```
./robot.sh han --oc --cmt -s
```

- Wait for the rule loading process to finish at tmux window 6

- Sometimes it crashes at this point, with a `Floating point exception(core dumped)`, after loading the rules

- If it hasn't, open a Chrome browser, go to `http://localhost:8000/#/interactions` and chat with it, it usually crashes after sending two to three messages to it.

- We tried to get the backtrace using gdb, but gdb stops responding after printing these:
```
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `guile -l btree-psi.scm'.
Program terminated with signal SIGFPE, Arithmetic exception.
#0  0x00007fb94e2c6685 in operator() (this=<optimized out>, __num=<optimized out>, 
    __den=<optimized out>) at /usr/include/c++/4.8/bits/hashtable_policy.h:345
345	    { return __num % __den; }
(gdb) bt
#0  0x00007fb94e2c6685 in operator() (this=<optimized out>, __num=<optimized out>, 
    __den=<optimized out>) at /usr/include/c++/4.8/bits/hashtable_policy.h:345
#1  _M_bucket_index (this=<optimized out>, __c=<optimized out>, __n=<optimized out>)
    at /usr/include/c++/4.8/bits/hashtable_policy.h:1023
#2  _M_bucket_index (this=<optimized out>, __k=<optimized out>, __c=<optimized out>)
    at /usr/include/c++/4.8/bits/hashtable.h:593
```

ngeiswei, 2016-09-05:
Could you describe how to reproduce that bug with the blender head?

leungmanhin, 2016-09-05:
The above steps can be used to reproduced the bug with the blender head, I mean, we used to use `dev.sh` in HEAD to run and develop things locally, but `dev.sh` only runs open sourced stuffs, and to run things on an actual robot we have to run some proprietary hansonrobotics stuffs as well, in which case the `hrtool` and `robot.sh` in `private_ws` handle that.

We can run `robot.sh` locally even without a physical robot, it will start a blender head and everything just like what `dev.sh` does, but the difference is it also loads the extra proprietary codes.

ngeiswei, 2016-09-05:
Clearly this operator is called with __den set to 0.

Still trying to reproduce that, I've tried
```
nilg@sofia-desktop:~/hansonrobotics/HEAD/scripts$ ./dev.sh han --oc --cmt -s
Unknown argument han
Usage: ./dev.sh [--cmt] [--oc] [--sa]
  --cmt
    Run CMT based face tracking
  --oc
    Run OpenCog based chatbot and behavior tree
  --sa
    Enable OpenCog NLP sentiment analysis, should run with --oc
  --st
    Enable saliency tracking.
```
I'll do some trials but the exact command line would be welcome nevertheless.

ngeiswei, 2016-09-05:
I can't reproduce that. The chat seems frozen after replying a few sentences though, but btree is still running.

ngeiswei, 2016-09-05:
Actually I managed to reproduce it. I disabled cmt (l.135 of dev.sh, cause this has always been a source of problems for me), and run `./dev.sh --oc`.

ngeiswei, 2016-09-05:
I also can't get the whole trace (gdb takes a full core but refuses to display more than the last calls), that is an immense pain. The fine logs don't seem to reveal anything in particular. I really don't know where to go from there! Has anybody the slightest clue?

ngeiswei, 2016-09-05:
This bug is fairly recent, does someone know which commits of ros-behavior-scripting and opencog worked before that?

linas, 2016-09-05:
Will look. An obvious first step is to disable CMT and/or the new spacemap code.  A much longer stack trace would help.  In theory, the longer stack trace can be found in the file `opencog.log` -- is it not being created or written to?

ngeiswei, 2016-09-06:
`opencog.log` leaves no trace.

I report my findings so far, neither atomspace nor opencog seem to be the problem.

## Test commits from the atomspace repo

https://github.com/opencog/atomspace/commit/fc594d617e63f5804b7cb9eb5e1b5c0e59506e19
https://github.com/opencog/atomspace/commit/0e5fa347ce7176e0bcb0a16957ed182ee3ebd525

have the same bug.

The following have other problems so I could not test

https://github.com/opencog/atomspace/commit/ed604ddcd3ac7d2a35ba372f1b02783bb2509e8a
https://github.com/opencog/atomspace/commit/8f4b79c1d09473900013ca938cd440e1bac8e1c0

## Test commits from the opencog repo

https://github.com/opencog/opencog/commit/227178765ef9d6b58ab7ec0671edea23ffe8dd5d
https://github.com/opencog/opencog/commit/eba9650e3fcd4329084aa7d25f7406ade39ffe95
https://github.com/opencog/opencog/commit/85437ccc1f79c57a2adb56100bf26b79b33c1a25

all have the bug.

Based on that I think the bug comes from the ros-behavior-scripting repo.

leungmanhin, 2016-09-06:
Yes this is happening with `--oc` only, and even without a webcam plugged in, and `opencog.log` leaves no trace...

Just throwing out ideas, besides code changes I've also updated the AIML rules being loaded in the past 1-2 weeks because the aiml2psi conversion script has been updated, so maybe somehow those updated rules trigger this problem?

The rules can be downloaded here: https://github.com/opencog/test-datasets/releases
- Currently it's loading [oc_public_aiml.tar.gz] (https://github.com/opencog/test-datasets/releases/download/current/oc_public_aiml.tar.gz)
- Older rules are still there with a name [aiml-current.tar.gz] (https://github.com/opencog/test-datasets/releases/download/current/aiml-current.tar.gz)

ngeiswei, 2016-09-06:
I've updated https://github.com/opencog/ros-behavior-scripting/issues/92#issuecomment-244848164 actually it's not obvious it comes from ros-behavior-scripting. Still searching...

leungmanhin, 2016-09-06:
I got the same error when tring to load only a rule file `generic-aiml-rules-Jul-12-2016.scm` that we used earlier

ngeiswei, 2016-09-06:
Guys, since I've updated ros-behavior-scripts with @AmeBel latest commits I've had 3x30min chat sessions with Sophia, could be just luck but I don't think so. I don't think the bug has been fixed it is probably just masked, but as far as demoing goes it should be OK for now.

linas, 2016-09-06:
I've hit at least half-a-dozen install bugs and issues, just trying to get to the point of running the thing.  The current one is this:
```
./robot.sh: line 120: /home/ubuntu/hansonrobotics/private_ws/devel/setup.bash: No such file or directory
```

linas, 2016-09-06:
Window 6 shows this: 
```
ERROR: In procedure apply-smob/1:
ERROR: Unbound variable: aiml-set-bot-prop
ABORT: unbound-variable
```

linas, 2016-09-06:
During build, I also get this:
```
Find octomap lib OCTOMAP_LIBRARY-NOTFOUND
Find octomath lib OCTOMAP_OCTOMATH_LIBRARY-NOTFOUND
```

linas, 2016-09-06:
Also this:
```
-- link-grammar( >= 4.7.12) was not found. Make sure LINK_GRAMMAR_LIBRARY, LINK_GRAMMAR_INCLUDE_DIR and LINK_GRAMMAR_DATA_DIR are set.
-- Link Grammar missing: needed for NLP.
```

linas, 2016-09-07:
And this:
```
./hansonrobotics/HEAD/scripts/hrtool: line 573: cd: /opt/hansonrobotics/vision/CppMT: No such file or directory
```

linas, 2016-09-07:
Terminal 1 is spewing immense numbers of errors. Unfortunately, because byobu wasn't used, the messages scrolled off the screen.  Why, exactly, does everyone avoid byobu?

wenwei-dev, 2016-09-07:
This is likely because hrtool -i was not successfully finished. CppMT is
one of the cmt tracker dependencies. If you use --cmt, it's needed.

wenwei-dev, 2016-09-07:
Unfortunately the CI for hrtool is down these days due to its migration
from scripts repository to private_ws/scripts. So hrtool is not tested.

Just wonder which repository are you using? The scripts should be
in private_ws/scripts, not the scripts repository itself.

linas, 2016-09-07:
Is there a way to tell if CMT is working?  It does not seem to open a feedback window, anywhere, and the behavior tree suggests that she never sees anyone.   The webcam is working -- I am able to run cheese, and see live video.  Also, pi-vision works, so somehow its just cmt that isn;'t doing anything?

linas, 2016-09-07:
blender crashed after a few hours of running. Chatbot has not crashed, despite a long stupid conversation.

leungmanhin, 2016-09-07:
after doing `hrtool -U`, it seems that the chatbot is no longer crashing, but it still crashes if it detects a face

ngeiswei, 2016-09-07:
Damn it, so I suppose you still need help on that front. The fact that we don't have the whole trace is a major pain, maybe I'll look into gdb to fix that...

ngeiswei, 2016-09-07:
Can't reproduce this problem even with cmt, with it I have other problems (like I've always have)
```
[rosout][ERROR] [1473240348.812848]: bad callback: <bound method FaceTrack.stt_cb of <face_track.FaceTrack instance at 0x7f3210710518>>
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/topics.py", line 720, in _invoke_callback
    cb(msg)
  File "/home/nilg/hansonrobotics/opencog/ros-behavior-scripting/face_track/face_track.py", line 399, in stt_cb
    self.atomo.who_said(msg.utterance)
  File "/home/nilg/hansonrobotics/opencog/ros-behavior-scripting/face_track/face_atomic.py", line 43, in who_said
    netcat(self.hostname, self.port, spoke + "\n")
  File "/home/nilg/hansonrobotics/opencog/ros-behavior-scripting/face_track/netcat.py", line 42, in netcat
    data = s.recv(1024)
timeout: timed out
```
But that's another issue.

leungmanhin, 2016-09-07:
We have tested it a few more times, sometimes it doesn't crash, sometimes it crashes with the floating point exception, sometimes it crashes with a segfault, I'll open another github issue for the segfault one

leungmanhin, 2016-09-07:
@ngeiswei for the error you mentioned, could you telnet to the cogserver and see if you can interact with the atomspace after getting the error?

ngeiswei, 2016-09-07:
No, I can't (though I'm not sure what the port is supposed to be, 17001?). However it seems related to this error in the cog window
```
Backtrace:
In ice-9/boot-9.scm:
 157: 2 [catch #t #<catch-closure 9c32b00> ...]
In unknown file:
   ?: 1 [apply-smob/1 #<catch-closure 9c32b00>]
   ?: 0 [opencog-extension cog-evaluate! ((DefinedPredicateNode "OpenPsi: loop")
)]

ERROR: In procedure opencog-extension:
ERROR: Throw to key `C++-EXCEPTION' with args `("cog-evaluate!" "Backtrace:\nIn ice-9/boot-9.scm:\n 157: 16 [catch #t #<catch-closure 9c32b00> ...]\nIn unknown file:\n   ?: 15 [apply-smob/1 #<catch-closure 9c32b00>]\n   ?: 14 [opencog-extension cog-evaluate! ((DefinedPredicateNode \"OpenPsi: loop\")\n)]\nIn ice-9/boot-9.scm:\n 157: 13 [catch #t #<catch-closure 7f64420> ...]\nIn unknown file:\n   ?: 12 [apply-smob/1 #<catch-closure 7f64420>]\nIn ice-9/boot-9.scm:\n 157: 11 [catch #t #<catch-closure 7f64300> ...]\nIn unknown file:\n   ?: 10 [apply-smob/1 #<catch-closure 7f64300>]\nIn opencog/openpsi/main.scm:\n  97: 9 [psi-step]\nIn srfi/srfi-1.scm:\n 576: 8 [map #<procedure 134faa0 at opencog/openpsi/main.scm:98:8 (d)> (# # # #)]\nIn opencog/openpsi/main.scm:\n 108: 7 [#<procedure 134faa0 at opencog/openpsi/main.scm:98:8 (d)> #]\nIn opencog/openpsi/action-selector.scm:\n 247: 6 [psi-default-action-selector-per-demand # #]\nIn unknown file:\n   ?: 5 [filter #<procedure 12d00a0 at opencog/openpsi/rule.scm:316:8 (x)> #]\nIn opencog/openpsi/rule.scm:\n 317: 4 [#<procedure 12d00a0 at opencog/openpsi/rule.scm:316:8 (x)> #]\n 290: 3 [psi-satisfiable? #]\nIn unknown file:\n   ?: 2 [opencog-extension cog-evaluate! (#)]\nIn ice-9/boot-9.scm:\n 102: 1 [#<procedure 84af240 at ice-9/boot-9.scm:97:6 (thrown-k . args)> C++-EXCEPTION ...]\nIn unknown file:\n   ?: 0 [apply-smob/1 #<catch-closure 7f642c0> C++-EXCEPTION ...]\n\nERROR: In procedure apply-smob/1:\nERROR: In procedure cog-evaluate!: Cannot call getMean() method of a NullTruthvalue (/home/nilg/hansonrobotics/opencog/atomspace/opencog/truthvalue/NullTruthValue.cc:52)\nABORT: C++-EXCEPTION\n (/home/nilg/hansonrobotics/opencog/atomspace/opencog/guile/SchemeEval.cc:1201)")'.
```

yantrabuddhi, 2016-09-07:
Hi Nil,
 The port being used now is 17020.

linas, 2016-09-08:
Behavior tree just halted for me,
```
Guile comes with ABSOLUTELY NO WARRANTY; for details type `,show w'.
This program is free software, and you are welcome to redistribute it
under certain conditions; type `,show c' for details.

Enter `,help' for help.
scheme@(guile-user)> --- Wake up! Elapsed: 1473293325.884853 seconds
----- Behavior pub: Waking up
Publish soma state: normal intensity: 0.1
Publish expression: happy
Published gesture:  shake-3
----- Behavior pub: This is boring
--- Bored! nothing is happening!
look at point:  1.02758256189 0.127575 0.209425538604
terminate called after throwing an instance of 'std::length_error'
  what():  basic_string::_S_create
/home/ubuntu/hansonrobotics/private_ws/scripts/btree.sh: line 20: 15342 Aborted                 $cmd
```

linas, 2016-09-08:
Nil, the error
```
Cannot call getMean() method of a NullTruthvalue
```
has for the rest of it, this stack trace:
```
psi-satisfiable?
opencog/openpsi/rule.scm:316
In opencog/openpsi/rule.scm:\n 317:
filter #<procedure 12d00a0 at opencog/openpsi/rule.scm:316
In opencog/openpsi/action-selector.scm:\n 247: 6 [psi-default-action-selector-per-demand
at opencog/openpsi/main.scm:98:8
In opencog/openpsi/main.scm:\n 108
at opencog/openpsi/main.scm:98
In opencog/openpsi/main.scm:\n  97
```

Crawling through this, I see that `opencog/openpsi/rule.scm:316` was presumably given a NullTruthValue.  One fix would be to make sure it has a valid TV by saying `(if (cog-tv? ...))` to be sure. However, it should not be getting bad TV's in any case. So where did this come from?  Well, I guess its `In opencog/openpsi/main.scm:\n 108`  which suggests that line 111 `(psi-get-all-valid-demands)` is returning demands that have NullTruthValues attached to them.  Confusing. Looking around some more.

linas, 2016-09-08:
Here's how the NullTruthValue got in there. Some old code returns `TruthValue::NULL_TV()` when there is an evaluation error, instead of throwing an exception.  Later on, this gets dereferenced.   I will change this shortly to throw an exception, instead of returning this bogus value.  The exception will result in some brand-new failure, but at least we will be able to find it.

linas, 2016-09-08:
Hmm, I have to take back part of the previous comment. The code that returns NullTruthValue() is "working as designed" -- its returning a bad value when an error occurred. An exception is already caught and logged, and I cannot pass that exception up to higher layers, as that breaks excpetion handling in GroundedPredicateNode, GroundedSchemaNode. 

So .. somewhere near opencog/openpsi/main.scm: line 108  -- the code that is supplying valid demands to open-psi -- somewhere there is an exception, which should be getting printed to stdout, or at least to the log file (?) -- that is causing a NullTruthValue to get returned.

Pull request opencog/atomspace#872 will allow you to use `cog-tv?` to test to see if the returned tv really is valid or not.

linas, 2016-09-21:
closing, believed to be fixed.

closed 2016-09-21 by linas