opencog/ros-behavior-scripting #93: EvaControl instance has no attribute 'running'

wenwei-dev, 2016-09-06:
It can't respond to "btree_on" behavior_switch command. Is there anything blocking the initialization of running attribute?

@yantrabuddhi This line looks suspicious: https://github.com/opencog/ros-behavior-scripting/blob/master/src/ros_commo.py#L527

```
[rosout][ERROR] [1473127414.176234]: bad callback: <bound method EvaControl.behavior_switch_callback of <ros_commo.EvaControl instance at 0x7fdf53be1b48>>
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/topics.py", line 720, in _invoke_callback
    cb(msg)
  File "ros_commo.py", line 375, in behavior_switch_callback
    if not self.running:
AttributeError: EvaControl instance has no attribute 'running'
```

linas, 2016-09-21:
is this still an issue?

AmeBel, 2016-09-26:
Just tested, behavior_switch is having the expected behavior on both btree-psi and btree

closed 2016-09-26 by AmeBel