opencog/ros-behavior-scripting #98: Crash with segfault

leungmanhin, 2016-09-07:
Similar to https://github.com/opencog/ros-behavior-scripting/issues/92, it sometimes segfault, this is the backtrace

```
(gdb) bt
#0  opencog::AtomTable::getHandle (this=0x7f24dcff7800, uuid=uuid@entry=18446744073709551615)
    at /usr/include/c++/4.8/bits/hashtable.h:1038
#1  0x00007f2642bf5f86 in do_res (uuid=<optimized out>)
    at /home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/atoms/base/Handle.cc:120
#2  opencog::Handle::Handle(unsigned long) (this=this@entry=0x7f2514ff5970, 
    u=18446744073709551615)
    at /home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/atoms/base/Handle.cc:39
#3  0x00007f26434f944e in opencog::SchemeSmob::ss_atom (suuid=<optimized out>)
    at /home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/guile/SchemeSmobNew.cc:199
#4  0x00007f2645ccd13f in ?? () from /usr/lib/libguile-2.0.so.22
#5  0x00007f2645c4fdc3 in scm_primitive_eval () from /usr/lib/libguile-2.0.so.22
#6  0x00007f2645c4fe23 in scm_eval () from /usr/lib/libguile-2.0.so.22
#7  0x00007f2645ccd12a in ?? () from /usr/lib/libguile-2.0.so.22
#8  0x00007f2645c4f863 in scm_call_4 () from /usr/lib/libguile-2.0.so.22
#9  0x00007f26434fbdbf in opencog::SchemeEval::do_scm_eval(scm_unused_struct*, scm_unused_struct* (*)(void*)) (this=this@entry=0x7f25600009d0, sexpr=0x712b530, 
    evo=0x7f2643502610 <thunk_scm_eval(void*) [clone .11153]>)
    at /home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/guile/SchemeEval.cc:833
#10 0x00007f26434fc209 in opencog::SchemeEval::do_apply_scm(std::string const&, opencog::Handle const&) (this=this@entry=0x7f25600009d0, func=..., varargs=...)
    at /home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/guile/SchemeEval.cc:1163
#11 0x00007f2643501c21 in opencog::SchemeEval::apply_tv (this=this@entry=0x7f25600009d0, 
    func=..., varargs=...)
    at /home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/guile/SchemeEval.cc:1181
#12 0x00007f263c482813 in opencog::EvaluationLink::do_evaluate (as=as@entry=0x8e6b10, pn=..., 
    cargs=...)
    at /home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/atoms/execution/EvaluationLink.cc:588
#13 0x00007f263c47664f in opencog::EvaluationLink::do_eval_scratch (as=<optimized out>, 
    evelnk=..., scratch=0x8e6b10, silent=silent@entry=true)
    at /home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/atoms/execution/EvaluationLink.cc:260
#14 0x00007f263c4825d0 in opencog::EvaluationLink::do_evaluate (as=<optimized out>, 
    evelnk=..., silent=silent@entry=true)
    at /home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/atoms/execution/EvaluationLink.cc:484
#15 0x00007f263c47532b in opencog::Instantiator::walk_tree (this=this@entry=0x7f2514ff6430, 
    expr=...)
    at /home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/atoms/execution/Instantiator.cc:227
#16 0x00007f263c482d52 in opencog::Instantiator::instantiate (this=this@entry=0x7f2514ff6430, 
    expr=..., vars=...)
    at /home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/atoms/execution/Instantiator.cc:477
#17 0x00007f263c482e7d in opencog::Instantiator::execute(opencog::Handle const&) [clone .local.665] (this=this@entry=0x7f2514ff6430, expr=...)
    at /home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/atoms/execution/Instantiator.h:101
#18 0x00007f263c477710 in opencog::EvaluationLink::do_eval_scratch (as=as@entry=0x8e6b10, 
    evelnk=..., scratch=scratch@entry=0x8e6b10, silent=silent@entry=false)
    at /home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/atoms/execution/EvaluationLin---Type <return> to continue, or q <return> to quit---
k.cc:404
#19 0x00007f263c476f87 in opencog::EvaluationLink::do_eval_scratch (as=as@entry=0x8e6b10, 
    evelnk=..., scratch=scratch@entry=0x8e6b10, silent=silent@entry=false)
    at /home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/atoms/execution/EvaluationLink.cc:315
#20 0x00007f263c4776b9 in opencog::EvaluationLink::do_eval_scratch (as=as@entry=0x8e6b10, 
    evelnk=..., scratch=scratch@entry=0x8e6b10, silent=silent@entry=false)
    at /home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/atoms/execution/EvaluationLink.cc:455
#21 0x00007f263c476f87 in opencog::EvaluationLink::do_eval_scratch (as=as@entry=0x8e6b10, 
    evelnk=..., scratch=scratch@entry=0x8e6b10, silent=silent@entry=false)
    at /home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/atoms/execution/EvaluationLink.cc:315
#22 0x00007f263c4776b9 in opencog::EvaluationLink::do_eval_scratch (as=0x8e6b10, evelnk=..., 
    scratch=0x8e6b10, silent=silent@entry=false)
    at /home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/atoms/execution/EvaluationLink.cc:455
#23 0x00007f263c4825d0 in opencog::EvaluationLink::do_evaluate (as=<optimized out>, 
    evelnk=..., silent=silent@entry=false)
    at /home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/atoms/execution/EvaluationLink.cc:484
#24 0x00007f263c48140b in ss_evaluate(opencog::AtomSpace*, opencog::Handle const&) [clone .93437] (atomspace=<optimized out>, h=...)
    at /home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/atoms/execution/ExecSCM.cc:41
#25 0x00007f2643503492 in opencog::FunctionWrap::as_wrapper_p_h (this=0x8e61e0, h=...)
    at /home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/guile/SchemeModule.cc:54
#26 0x00007f26434f4e7c in opencog::SchemePrimitive<opencog::FunctionWrap>::invoke(scm_unused_struct*) [clone .local.289] (this=0x11402f0, args=<optimized out>)
    at /home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/guile/SchemePrimitive.h:634
#27 0x00007f26435068f8 in opencog::PrimitiveEnviron::do_call (sfe=0x191fe40, 
    arglist=0x712b560)
[... 84 lines cut ...]
#69 0x00007f263c48140b in ss_evaluate(opencog::AtomSpace*, opencog::Handle const&) [clone .93437] (atomspace=<optimized out>, h=...)
    at /home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/atoms/execution/ExecSCM.cc:41
#70 0x00007f2643503492 in opencog::FunctionWrap::as_wrapper_p_h (this=0x8e61e0, h=...)
    at /home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/guile/SchemeModule.cc:54
#71 0x00007f26434f4e7c in opencog::SchemePrimitive<opencog::FunctionWrap>::invoke(scm_unused_struct*) [clone .local.289] (this=0x11402f0, args=<optimized out>)
    at /home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/guile/SchemePrimitive.h:634
#72 0x00007f26435068f8 in opencog::PrimitiveEnviron::do_call (sfe=0x191fe40, 
    arglist=0x790f0f0)
    at /home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/guile/SchemePrimitive.cc:150
#73 0x00007f2645ccd12a in ?? () from /usr/lib/libguile-2.0.so.22
#74 0x00007f2645cbff80 in ?? () from /usr/lib/libguile-2.0.so.22
#75 0x00007f2645c462ca in ?? () from /usr/lib/libguile-2.0.so.22
#76 0x00007f2645ccd12a in ?? () from /usr/lib/libguile-2.0.so.22
#77 0x00007f2645c4f863 in scm_call_4 () from /usr/lib/libguile-2.0.so.22
#78 0x00007f2645c469ff in ?? () from /usr/lib/libguile-2.0.so.22
---Type <return> to continue, or q <return> to quit---
#79 0x00007f2645c46a95 in scm_c_with_continuation_barrier () from /usr/lib/libguile-2.0.so.22
#80 0x00007f2645cbf78c in ?? () from /usr/lib/libguile-2.0.so.22
#81 0x00007f26453b4fe2 in GC_call_with_stack_base () from /usr/lib/x86_64-linux-gnu/libgc.so.1
#82 0x00007f2645cbf14c in ?? () from /usr/lib/libguile-2.0.so.22
#83 0x00007f26453b9d93 in GC_inner_start_routine () from /usr/lib/x86_64-linux-gnu/libgc.so.1
#84 0x00007f26453b4fe2 in GC_call_with_stack_base () from /usr/lib/x86_64-linux-gnu/libgc.so.1
#85 0x00007f26459d5182 in start_thread (arg=0x7f2514ff9700) at pthread_create.c:312
#86 0x00007f264570247d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
```

linas, 2016-09-08:
OK, so I see one issue here -- the call `cog-atom` is being used, and worse -- it is being passed an undefined value.  The code is here: 
```
$ grep -r cog-atom *
src/time-map.scm:		(if (equal? (cog-atom (cog-undefined-handle)) loc-atom)
src/time-map.scm:			(if (equal? (cog-atom (cog-undefined-handle)) loc-atom)
src/time-map.scm:		(if (equal? (cog-atom (cog-undefined-handle)) loc-atom)
src/time-map.scm:		(if (equal? (cog-atom (cog-undefined-handle)) loc-atom)
```

which is fundamentally just-plain bad code. I will fix this shortly.

Of course, bad cod should not make things crash, either, so I will add a check there as well.

linas, 2016-09-08:
OK so pull request opencog/atomspace#870 should avoid the crash, at least, by throwing an error, instead.

linas, 2016-09-08:
OK, so pull request #100 should provide the other half of that fix -- it replaces the usage 
```
(if (equal? (cog-atom (cog-undefined-handle)) loc-atom)
```
with
```
(if (not (cog-atom? loc-atom)))
```
which is the right way to test for invalid atoms.

linas, 2016-09-08:
p.s. I was not able to actually test the fix --  I still do not have a working face-tracker in the current hanson-robotic code base.  However, this should take care of this particular crash.

linas, 2016-09-21:
closing I believe this is fixed

closed 2016-09-21 by linas