opencog/ros-behavior-scripting #99: Another segfault

leungmanhin, 2016-09-07:
Similar to https://github.com/opencog/ros-behavior-scripting/issues/98 but with a different backtrace, gdb stops responding after printing these:
```
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `guile -l btree-psi.scm'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007feedcc7c68d in std::_Hashtable<unsigned long, std::pair<unsigned long const, opencog::Handle>, std::allocator<std::pair<unsigned long const, opencog::Handle> >, std::__detail::_Select1st, std::equal_to<unsigned long>, std::hash<unsigned long>, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<false, false, true> >::_M_find_before_node(unsigned long, unsigned long const&, unsigned long) const [clone .isra.448] (__k=<optimized out>, __n=<optimized out>, 
    this=<optimized out>) at /usr/include/c++/4.8/bits/hashtable.h:1159
1159	      __node_base* __prev_p = _M_buckets[__n];
(gdb) bt
#0  0x00007feedcc7c68d in std::_Hashtable<unsigned long, std::pair<unsigned long const, opencog::Handle>, std::allocator<std::pair<unsigned long const, opencog::Handle> >, std::__detail::_Select1st, std::equal_to<unsigned long>, std::hash<unsigned long>, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<false, false, true> >::_M_find_before_node(unsigned long, unsigned long const&, unsigned long) const [clone .isra.448] (__k=<optimized out>, __n=<optimized out>, 
    this=<optimized out>) at /usr/include/c++/4.8/bits/hashtable.h:1159
#1  _M_find_node (this=<optimized out>, __bkt=<optimized out>, __key=<optimized out>, 
    __c=<optimized out>) at /usr/include/c++/4.8/bits/hashtable.h:604
```

linas, 2016-09-08:
Pull request opencog/atomspace#871 should fix this. 

I think its really weird, that a find of -1 should cause a crash here, but it appears that passing a -1 in, as an unsigned long const, to the stdc hashing function is ... bad.  This makes me think that maybe there is a bug in /usr/include/c++/4.8/bits/hashtable.h  which seems very unlikely, but not sure how else to explain this.  Another possibility is that RAM was already corrupted, before this point.

Anyway, this fix, plus the fixes in bug #98 should prevent thisfrom happening.

linas, 2016-09-21:
closing, I believe this is fixed

closed 2016-09-21 by linas