opencog/ros-behavior-scripting #101: Yet another segfault

leungmanhin, 2016-09-08:
From the latest code today, I get another segfault...
```
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `guile -l btree-psi.scm'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  opencog::AtomTable::getHandle (this=0x7f1774ff7800, uuid=uuid@entry=0)
    at /usr/include/c++/4.8/bits/hashtable.h:1038
1038	    find(const key_type& __k) const
(gdb) bt
#0  opencog::AtomTable::getHandle (this=0x7f1774ff7800, uuid=uuid@entry=0)
    at /usr/include/c++/4.8/bits/hashtable.h:1038
#1  0x00007f186338ef96 in do_res (uuid=0)
    at /home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/atoms/base/Handle.cc:121
#2  opencog::Handle::Handle(unsigned long) (this=0x7f16e8079ae0, u=0)
    at /home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/atoms/base/Handle.cc:39
#3  0x00007f1852500170 in octomap::OccupancyOcTreeBase<octomap::AtomOcTreeNode>::updateNode (
    this=0x6e5f000, key=..., log_odds_update=0.847297847, lazy_eval=<optimized out>)
    at /home/leungmanhin/hansonrobotics/opencog/opencog/opencog/timeoctomap/AtomOcTreeNode.h:46
#4  0x00007f18524fc88b in octomap::OccupancyOcTreeBase<octomap::AtomOcTreeNode>::updateNode (
    this=0x6e5f000, value=..., occupied=occupied@entry=true, lazy_eval=lazy_eval@entry=false)
    at /opt/ros/indigo/include/octomap/OccupancyOcTreeBase.hxx:354
#5  0x00007f1852503fb8 in TimeOctomap::put_atom_at_current_time (this=0x5a10ff0, 
    location=..., ato=...)
    at /home/leungmanhin/hansonrobotics/opencog/opencog/opencog/timeoctomap/TimeOctomap.cc:154
#6  0x00007f185271e5ac in opencog::ato::PointMemorySCM::map_ato (this=<optimized out>, 
    map_name=..., ato=..., x=<optimized out>, y=<optimized out>, z=<optimized out>)
    at /home/leungmanhin/hansonrobotics/opencog/opencog/opencog/timeoctomap/pointmemory/PointMemorySCM.cc:177
#7  0x00007f1852727dbd in opencog::SchemePrimitive<opencog::ato::PointMemorySCM>::invoke (
    this=0x208e330, args=<optimized out>)
    at /usr/local/include/opencog/guile/SchemePrimitive.h:283
#8  0x00007f1863c9f928 in opencog::PrimitiveEnviron::do_call (sfe=0x1e68480, 
    arglist=0xc0373a0)
    at /home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/guile/SchemePrimitive.cc:150
#9  0x00007f186646612a in ?? () from /usr/lib/libguile-2.0.so.22
#10 0x00007f18663e8747 in scm_call_1 () from /usr/lib/libguile-2.0.so.22
#11 0x00007f186646612a in ?? () from /usr/lib/libguile-2.0.so.22
#12 0x00007f18663e881e in scm_call_3 () from /usr/lib/libguile-2.0.so.22
#13 0x00007f186646612a in ?? () from /usr/lib/libguile-2.0.so.22
#14 0x00007f18663e8863 in scm_call_4 () from /usr/lib/libguile-2.0.so.22
#15 0x00007f1863c9dea1 in opencog::SchemeEval::do_eval (this=0x7f16fc1d3470, expr=...)
    at /home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/guile/SchemeEval.cc:655
#16 0x00007f1863c9dfc3 in opencog::SchemeEval::c_wrap_eval (p=0x7f16fc1d3470)
    at /home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/guile/SchemeEval.cc:583
#17 0x00007f18663df2ca in ?? () from /usr/lib/libguile-2.0.so.22
#18 0x00007f186646612a in ?? () from /usr/lib/libguile-2.0.so.22
#19 0x00007f18663e8863 in scm_call_4 () from /usr/lib/libguile-2.0.so.22
#20 0x00007f18663df9ff in ?? () from /usr/lib/libguile-2.0.so.22
#21 0x00007f18663dfa95 in scm_c_with_continuation_barrier () from /usr/lib/libguile-2.0.so.22
#22 0x00007f1865b53950 in GC_call_with_gc_active () from /usr/lib/x86_64-linux-gnu/libgc.so.1
#23 0x00007f18664587d1 in ?? () from /usr/lib/libguile-2.0.so.22
#24 0x00007f1865b4dfe2 in GC_call_with_stack_base () from /usr/lib/x86_64-linux-gnu/libgc.so.1
#25 0x00007f1866458b48 in scm_with_guile () from /usr/lib/libguile-2.0.so.22
#26 0x00007f1863c9df6e in opencog::SchemeEval::eval_expr (this=0x7f16fc1d3470, expr=...)
    at /home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/guile/SchemeEval.cc:551
#27 0x00007f1862eeba60 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#28 0x00007f186616e182 in start_thread (arg=0x7f184707f700) at pthread_create.c:312
#29 0x00007f1865e9b47d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
```

linas, 2016-09-08:
OK, I want to get some sleep, but I think that tracking this down should be easy: line 46 of timeoctomap/AtomOcTreeNode.h initializes a handle to a uuid of zero.  Just don't do that, do this:
```
$ git diff
diff --git a/opencog/timeoctomap/AtomOcTreeNode.h b/opencog/timeoctomap/AtomOcTr
index b4c85dc..f76969d 100644
--- a/opencog/timeoctomap/AtomOcTreeNode.h
+++ b/opencog/timeoctomap/AtomOcTreeNode.h
@@ -43,7 +43,7 @@ const opencog::Handle UndefinedHandle = opencog::Handle::UNDEF
 class AtomOcTreeNode : public OcTreeNode
 {
 public:
-    AtomOcTreeNode() : OcTreeNode(), dat(0)
+    AtomOcTreeNode() : OcTreeNode()
     {} //dat gets default value from prunning
 
     AtomOcTreeNode(const AtomOcTreeNode& rhs) : OcTreeNode(rhs), dat(rhs.dat)
```

and that should fix things.

ngeiswei, 2016-09-08:
I'll look into it.

ngeiswei, 2016-09-08:
Crap, can't reproduce that, instead getting this incomplete (understatement) trace
```
#0  std::_Hashtable<unsigned long, std::pair<unsigned long const, opencog::Handle>, std::allocator<std::pair<unsigned long const, opencog::Handle> >, std::__detail::_Select1st, std::equal_to<unsigned long>, std::hash<unsigned long>, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<false, false, true> >::_M_find_before_node(unsigned long, unsigned long const&, unsigned long) const [clone .isra.448] (
```
gdb freezes again, this is a major pain (sounds more like a minor to my ears).

ngeiswei, 2016-09-08:
Tested for about 30min without crashing (including using cmt). Thanks Linas!!!!

leungmanhin, 2016-09-08:
Have been running it for an hour and not getting any segfault, very cool, thanks Linas!

linas, 2016-09-21:
closing, believed to be fixed

closed 2016-09-21 by linas