opencog/ros-behavior-scripting #102: Crash with an exception

leungmanhin, 2016-09-08:
It eventually crashed, with this:
```
Backtrace:
In ice-9/boot-9.scm:
 157: 2 [catch #t #<catch-closure 969d5a0> ...]
In unknown file:
   ?: 1 [apply-smob/1 #<catch-closure 969d5a0>]
   ?: 0 [opencog-extension cog-evaluate! ((DefinedPredicateNode "OpenPsi: loop")
)]

ERROR: In procedure opencog-extension:
ERROR: Throw to key `C++-EXCEPTION' with args `("cog-evaluate!" "Backtrace:\nIn ice-9/boot-9.scm:\n 157: 13 [catch #t #<catch-closure 969d5a0> ...]\nIn unknown file:\n   ?: 12 [apply-smob/1 #<catch-closure 969d5a0>]\n   ?: 11 [opencog-extension cog-evaluate! ((DefinedPredicateNode \"OpenPsi: loop\")\n)]\nIn ice-9/boot-9.scm:\n 157: 10 [catch #t #<catch-closure 98091a0> ...]\nIn unknown file:\n   ?: 9 [apply-smob/1 #<catch-closure 98091a0>]\nIn ice-9/boot-9.scm:\n 157: 8 [catch #t #<catch-closure 9809080> ...]\nIn unknown file:\n   ?: 7 [apply-smob/1 #<catch-closure 9809080>]\nIn opencog/openpsi/main.scm:\n  97: 6 [psi-step]\nIn srfi/srfi-1.scm:\n 578: 5 [map #<procedure 3398560 at opencog/openpsi/main.scm:98:8 (d)> (# # # #)]\n 578: 4 [map #<procedure act-and-evaluate (rule)> (#)]\nIn opencog/openpsi/main.scm:\n  85: 3 [act-and-evaluate #]\nIn unknown file:\n   ?: 2 [opencog-extension cog-execute! (#)]\nIn ice-9/boot-9.scm:\n 102: 1 [#<procedure 93ea140 at ice-9/boot-9.scm:97:6 (thrown-k . args)> C++-EXCEPTION ...]\nIn unknown file:\n   ?: 0 [apply-smob/1 #<catch-closure 9809040> C++-EXCEPTION ...]\n\nERROR: In procedure apply-smob/1:\nERROR: In procedure cog-execute!: Failed evaluation; see logfile for stack trace. (/home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/atoms/execution/ExecutionOutputLink.cc:159)\nABORT: C++-EXCEPTION\n (/home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/guile/SchemeEval.cc:1201)")'.
```

and from `opencog.log`:
```
[2016-09-08 10:24:08:842] [WARN] ServerSocket::Send(): Socket operation on non-socket on thread 0x7afd7700
[2016-09-08 10:24:08:842] [WARN] ServerSocket::Send(): Socket operation on non-socket on thread 0x7afd7700
[2016-09-08 11:01:54:184] [ERROR] Failed evaluation; see logfile for stack trace. (/home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/atoms/execution/ExecutionOutputLink.cc:159)
	Stack Trace:
	2: /usr/local/lib/libcogutil.so (opencog::Logger::logva(opencog::Logger::Level, char const*, __va_list_tag*) +0xa7) [0x7fa599b2b457]
	3: /usr/local/lib/libcogutil.so (opencog::Logger::Error::operator()(char const*, ...) +0x92) [0x7fa599b2b5f2]
	4: /usr/local/lib/libcogutil.so (opencog::StandardException::parse_error_message(char const*, __va_list_tag*, bool) +0x63) [0x7fa599b1d663]
	5: /usr/local/lib/libcogutil.so (opencog::StandardException::parse_error_message(char const*, char const*, __va_list_tag*, bool) +0x9a) [0x7fa599b1d73a]
	6: /usr/local/lib/libcogutil.so (opencog::RuntimeException::RuntimeException(char const*, char const*, ...) +0xb4) [0x7fa599b1d854]
	7: /usr/local/lib/opencog/libexecution.so (opencog::ExecutionOutputLink::do_execute(opencog::AtomSpace*, opencog::Handle const&, opencog::Handle const&) +0xad3) [0x7fa58b5eaf03]
	8: /usr/local/lib/opencog/libexecution.so (opencog::ExecutionOutputLink::execute(opencog::AtomSpace*) const +0x1d) [0x7fa58b5eb09d]
	9: /usr/local/lib/opencog/libexecution.so (opencog::Instantiator::walk_tree(opencog::Handle const&) +0xd2d) [0x7fa58b5e128d]
	10: /usr/local/lib/opencog/libexecution.so (opencog::Instantiator::walk_tree(opencog::Handle const&) +0xd8f) [0x7fa58b5e12ef]
	11: /usr/local/lib/opencog/libexecution.so (opencog::Instantiator::walk_tree(opencog::Handle const&) +0x739) [0x7fa58b5e0c99]
	12: /usr/local/lib/opencog/libexecution.so (opencog::Instantiator::instantiate(opencog::Handle const&, std::map<opencog::Handle, opencog::Handle, std::less<opencog::Handle>, std::allocator<std::pair<opencog::Handle const, opencog::Handle> > > const&) +0x32) [0x7fa58b5eed52]
	13: /usr/local/lib/opencog/libexecution.so (x +0x19e7d) [0x7fa58b5eee7d]
	14: /usr/local/lib/opencog/libexecution.so (opencog::EvaluationLink::do_eval_scratch(opencog::AtomSpace*, opencog::Handle const&, opencog::AtomSpace*, bool) +0x1420) [0x7fa58b5e3710]
	15: /usr/local/lib/opencog/libexecution.so (opencog::EvaluationLink::do_eval_scratch(opencog::AtomSpace*, opencog::Handle const&, opencog::AtomSpace*, bool) +0xc97) [0x7fa58b5e2f87]
	16: /usr/local/lib/opencog/libexecution.so (opencog::EvaluationLink::do_eval_scratch(opencog::AtomSpace*, opencog::Handle const&, opencog::AtomSpace*, bool) +0x13c9) [0x7fa58b5e36b9]
	17: /usr/local/lib/opencog/libexecution.so (opencog::EvaluationLink::do_evaluate(opencog::AtomSpace*, opencog::Handle const&, bool) +0x10) [0x7fa58b5ee5d0]
	18: /usr/local/lib/opencog/libexecution.so (x +0x1840b) [0x7fa58b5ed40b]
	19: /usr/local/lib/opencog/libsmob.so (opencog::FunctionWrap::as_wrapper_p_h(opencog::Handle) +0x22) [0x7fa599fcc542]
	20: /usr/local/lib/opencog/libsmob.so (x +0xde5c) [0x7fa599fbde5c]
	21: /usr/local/lib/opencog/libsmob.so (opencog::PrimitiveEnviron::do_call(scm_unused_struct*, scm_unused_struct*) +0x28) [0x7fa599fcf9a8]
	22: /usr/lib/libguile-2.0.so.22 (x +0xe212a) [0x7fa59c79612a]
	23: /usr/lib/libguile-2.0.so.22 (x +0x33) [0x7fa59c718dc3]
	24: /usr/lib/libguile-2.0.so.22 (x +0x53) [0x7fa59c718e23]
	25: /usr/lib/libguile-2.0.so.22 (x +0xe212a) [0x7fa59c79612a]
	26: /usr/lib/libguile-2.0.so.22 (x +0x33) [0x7fa59c718863]
	27: /usr/local/lib/opencog/libsmob.so (opencog::SchemeEval::do_scm_eval(scm_unused_struct*, scm_unused_struct* (*)(void*)) +0xbf) [0x7fa599fc5daf]
	28: /usr/local/lib/opencog/libsmob.so (opencog::SchemeEval::do_apply(std::string const&, opencog::Handle const&) +0x12) [0x7fa599fc6212]
	29: /usr/local/lib/opencog/libsmob.so (opencog::SchemeEval::apply(std::string const&, opencog::Handle) +0xd5) [0x7fa599fc6305]
	30: /usr/local/lib/opencog/libexecution.so (opencog::ExecutionOutputLink::do_execute(opencog::AtomSpace*, opencog::Handle const&, opencog::Handle const&) +0x55d) [0x7fa58b5ea98d]
	31: /usr/local/lib/opencog/libexecution.so (opencog::ExecutionOutputLink::execute(opencog::AtomSpace*) const +0x1d) [0x7fa58b5eb09d]
	32: /usr/local/lib/opencog/libexecution.so (opencog::Instantiator::walk_tree(opencog::Handle const&) +0xd2d) [0x7fa58b5e128d]
	33: /usr/local/lib/opencog/libexecution.so (opencog::Instantiator::walk_tree(opencog::Handle const&) +0xd8f) [0x7fa58b5e12ef]
	34: /usr/local/lib/opencog/libexecution.so (opencog::Instantiator::instantiate(opencog::Handle const&, std::map<opencog::Handle, opencog::Handle, std::less<opencog::Handle>, std::allocator<std::pair<opencog::Handle const, opencog::Handle> > > const&) +0x32) [0x7fa58b5eed52]
	35: /usr/local/lib/opencog/libexecution.so (x +0x19e7d) [0x7fa58b5eee7d]
	36: /usr/local/lib/opencog/libexecution.so (x +0x1799e) [0x7fa58b5ec99e]
	37: /usr/local/lib/opencog/libsmob.so (opencog::FunctionWrap::as_wrapper_h_h(opencog::Handle) +0x22) [0x7fa599fcc4d2]
	38: /usr/local/lib/opencog/libsmob.so (x +0xedb2) [0x7fa599fbedb2]
	39: /usr/local/lib/opencog/libsmob.so (opencog::PrimitiveEnviron::do_call(scm_unused_struct*, scm_unused_struct*) +0x28) [0x7fa599fcf9a8]
	40: /usr/lib/libguile-2.0.so.22 (x +0xe212a) [0x7fa59c79612a]
	41: /usr/lib/libguile-2.0.so.22 (x +0x33) [0x7fa59c718dc3]
	42: /usr/lib/libguile-2.0.so.22 (x +0x53) [0x7fa59c718e23]
	43: /usr/lib/libguile-2.0.so.22 (x +0xe212a) [0x7fa59c79612a]
	44: /usr/lib/libguile-2.0.so.22 (x +0x33) [0x7fa59c718863]
	45: /usr/local/lib/opencog/libsmob.so (opencog::SchemeEval::do_scm_eval(scm_unused_struct*, scm_unused_struct* (*)(void*)) +0xbf) [0x7fa599fc5daf]
	46: /usr/local/lib/opencog/libsmob.so (opencog::SchemeEval::c_wrap_apply_tv(void*) +0x19) [0x7fa599fcb899]
	47: /usr/lib/libguile-2.0.so.22 (x +0x5b2ca) [0x7fa59c70f2ca]
	48: /usr/lib/libguile-2.0.so.22 (x +0xe212a) [0x7fa59c79612a]
	49: /usr/lib/libguile-2.0.so.22 (x +0x33) [0x7fa59c718863]

[2016-09-08 11:01:54:335] [ERROR] Failed evaluation; see logfile for stack trace. (/home/leungmanhin/hansonrobotics/opencog/atomspace/opencog/atoms/execution/ExecutionOutputLink.cc:159)
[... 90 lines cut ...]
	8: /usr/local/lib/opencog/libexecution.so (opencog::EvaluationLink::do_evaluate(opencog::AtomSpace*, opencog::Handle const&, opencog::Handle const&) +0x233) [0x7fa58b5ee813]
	9: /usr/local/lib/opencog/libexecution.so (opencog::EvaluationLink::do_eval_scratch(opencog::AtomSpace*, opencog::Handle const&, opencog::AtomSpace*, bool) +0x35f) [0x7fa58b5e264f]
[... cut ...]
	24: /usr/lib/libguile-2.0.so.22 (x +0x35) [0x7fa59c70fa95]
	25: /usr/lib/libguile-2.0.so.22 (x +0xd478c) [0x7fa59c78878c]
	26: /usr/lib/x86_64-linux-gnu/libgc.so.1 (x +0x12) [0x7fa59be7dfe2]
	27: /usr/lib/libguile-2.0.so.22 (x +0xd414c) [0x7fa59c78814c]
	28: /usr/lib/x86_64-linux-gnu/libgc.so.1 (x +0x53) [0x7fa59be82d93]
	29: /usr/lib/x86_64-linux-gnu/libgc.so.1 (x +0x12) [0x7fa59be7dfe2]
	30: /lib/x86_64-linux-gnu/libpthread.so.0 (x +0x8182) [0x7fa59c49e182]
	31: /lib/x86_64-linux-gnu/libc.so.6 (x +0x6d) [0x7fa59c1cb47d]
```

linas, 2016-09-09:
Well, the appearence of the apply_tv and the trace pointing to opencog/openpsi/main.scm:97 and line 85 too suggests that this is the same NullTruthValue dereference, as discussed in #92   

Unfortunately, the way the logging is working, the original stack trace showing the NullTruthValue issue is not being printed/logged.  I will ponder some more about how to fix this up correctly.

linas, 2016-09-09:
OK, so I am pursuing multiple avenues:  one way to avoid this crash is to do this:
```
--- a/opencog/openpsi/rule.scm
+++ b/opencog/openpsi/rule.scm
@@ -313,8 +313,11 @@ there are 100K rules!
   have a non-zero strength
 "
     (filter
-        (lambda (x) (and (> (cog-stv-strength x) 0)
-            (equal? (stv 1 1) (psi-satisfiable? x))))
+        (lambda (x)
+            (and
+                (cog-tv? x)
+                (> (cog-stv-strength x) 0)
+                (equal? (stv 1 1) (psi-satisfiable? x))))
         (psi-get-rules demand-node))
 )
 
```
However, I am about to commit changes that will never actually pass a NullTV back to the user ... so ...

linas, 2016-09-09:
Interesting. The above `(cog-tv? x)` causes OpenPsiUTest to fail. So I think I can now hunt this down.

linas, 2016-09-09:
Never mind.  It should have been this:
```
(cog-tv? (cog-tv x))
```
Sorry for thinking aloud.

linas, 2016-09-09:
Pull req opencog/atomspace#875 should help clarify where the actual issue is occurring ... the bug should result with stack traces at the actual point of failure.

linas, 2016-09-21:
closing, believed to be fixed

closed 2016-09-21 by linas