opencog/ros-behavior-scripting #108: Segmentation fault when attempting to interact with the REST server is made

misgeatgit, 2016-09-20:
A segfault is happening when attempting to interact with the REST server in the private_ws hanson stack. Here is how to reproduce it.
-  Run the private_ws hanson stack
-  Enter to the python shell and load the restapi server  by typing the following commands

 ```python
import web.api.restapi
web.api.restapi.Start().run(None, ATOMSPACE)
```
-  Now, go to your browser or other http client and type the following GET request

````
   http://localhost:5000/api/v1.1/atoms? 
````
OR anyother supported request and you should see a segfault. Below is a stack trace of the segfault.

````
#0  0x00007fa5da75c8c5 in opencog::AtomTable::getHandlesByType<std::back_insert_iterator<std::vector<opencog::Handle, std::allocator<opencog::Handle> > > >(std::back_insert_iterator<std::vector<opencog::Handle, std::allocator<opencog::Handle> > >, unsigned short, bool, bool) const [clone .local.32] (this=0x0, result=..., result@entry=..., type=type@entry=4, subclass=<optimized out>, 
    parent=parent@entry=true) at /home/mandeep/hr/opencog/atomspace/opencog/atomspace/AtomTable.h:220
#1  0x00007fa5da75319c in get_handles_by_type (this=<optimized out>, result=..., type=<optimized out>, subclass=<optimized out>) at /home/mandeep/hr/opencog/atomspace/opencog/atomspace/AtomSpace.h:494
#2  __pyx_pf_7opencog_9atomspace_9AtomSpace_30get_atoms_by_type (__pyx_v_subtype=<optimized out>, __pyx_v_t=4, __pyx_v_self=0x7fa5e803a0b0)
    at /home/mandeep/hr/opencog/atomspace/build/opencog/cython/opencog/atomspace.cpp:5682
#3  __pyx_pw_7opencog_9atomspace_9AtomSpace_31get_atoms_by_type(_object*, _object*, _object*) [clone .14797] (__pyx_v_self=0x7fa5e803a0b0, __pyx_args=<optimized out>, __pyx_kwds=<optimized out>)
    at /home/mandeep/hr/opencog/atomspace/build/opencog/cython/opencog/atomspace.cpp:5646
#4  0x00007fa5df7f30d4 in PyEval_EvalFrameEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#5  0x00007fa5df7f454d in PyEval_EvalCodeEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#6  0x00007fa5df7f2dd8 in PyEval_EvalFrameEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#7  0x00007fa5df7f454d in PyEval_EvalCodeEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#8  0x00007fa5df8297a5 in ?? () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#9  0x00007fa5df795d43 in PyObject_Call () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#10 0x00007fa5df7ef3b1 in PyEval_EvalFrameEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#11 0x00007fa5df7f454d in PyEval_EvalCodeEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#12 0x00007fa5df8297a5 in ?? () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#13 0x00007fa5df795d43 in PyObject_Call () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#14 0x00007fa5df7ef3b1 in PyEval_EvalFrameEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#15 0x00007fa5df7f454d in PyEval_EvalCodeEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#16 0x00007fa5df8297a5 in ?? () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#17 0x00007fa5df795d43 in PyObject_Call () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#18 0x00007fa5df7ef3b1 in PyEval_EvalFrameEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#19 0x00007fa5df7f454d in PyEval_EvalCodeEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#20 0x00007fa5df8297a5 in ?? () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#21 0x00007fa5df795d43 in PyObject_Call () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#22 0x00007fa5df7ef3b1 in PyEval_EvalFrameEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#23 0x00007fa5df7f454d in PyEval_EvalCodeEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#24 0x00007fa5df8297a5 in ?? () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#25 0x00007fa5df795d43 in PyObject_Call () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#26 0x00007fa5df7ef3b1 in PyEval_EvalFrameEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#27 0x00007fa5df7f3059 in PyEval_EvalFrameEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#28 0x00007fa5df7f3059 in PyEval_EvalFrameEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#29 0x00007fa5df7f3059 in PyEval_EvalFrameEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#30 0x00007fa5df7f454d in PyEval_EvalCodeEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#31 0x00007fa5df8296d0 in ?? () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#32 0x00007fa5df795d43 in PyObject_Call () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#33 0x00007fa5df7217bd in ?? () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#34 0x00007fa5df795d43 in PyObject_Call () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#35 0x00007fa5df79a7e5 in ?? () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#36 0x00007fa5df795d43 in PyObject_Call () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#37 0x00007fa5df7f0816 in PyEval_EvalFrameEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#38 0x00007fa5df7f454d in PyEval_EvalCodeEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#39 0x00007fa5df7f2dd8 in PyEval_EvalFrameEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#40 0x00007fa5df7f454d in PyEval_EvalCodeEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#41 0x00007fa5df7f2dd8 in PyEval_EvalFrameEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#42 0x00007fa5df7f454d in PyEval_EvalCodeEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#43 0x00007fa5df7f2dd8 in PyEval_EvalFrameEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#44 0x00007fa5df7f454d in PyEval_EvalCodeEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#45 0x00007fa5df8296d0 in ?? () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#46 0x00007fa5df795d43 in PyObject_Call () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#47 0x00007fa5df7217bd in ?? () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#48 0x00007fa5df795d43 in PyObject_Call () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#49 0x00007fa5df7f0816 in PyEval_EvalFrameEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#50 0x00007fa5df7f454d in PyEval_EvalCodeEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
[... 26 lines cut ...]
#77 0x00007fa5df7ef3b1 in PyEval_EvalFrameEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#78 0x00007fa5df7f454d in PyEval_EvalCodeEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#79 0x00007fa5df7f2dd8 in PyEval_EvalFrameEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#80 0x00007fa5df7f454d in PyEval_EvalCodeEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#81 0x00007fa5df7f2dd8 in PyEval_EvalFrameEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#82 0x00007fa5df7f454d in PyEval_EvalCodeEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#83 0x00007fa5df8297a5 in ?? () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#84 0x00007fa5df795d43 in PyObject_Call () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#85 0x00007fa5df7ef3b1 in PyEval_EvalFrameEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#86 0x00007fa5df7f3059 in PyEval_EvalFrameEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#87 0x00007fa5df7f3059 in PyEval_EvalFrameEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#88 0x00007fa5df7f454d in PyEval_EvalCodeEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#89 0x00007fa5df8296d0 in ?? () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#90 0x00007fa5df795d43 in PyObject_Call () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#91 0x00007fa5df7217bd in ?? () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#92 0x00007fa5df795d43 in PyObject_Call () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#93 0x00007fa5df80e577 in PyEval_CallObjectWithKeywords () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#94 0x00007fa5df776f92 in ?? () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#95 0x00007fa5f1056182 in start_thread (arg=0x7fa4ab7f6700) at pthread_create.c:312
#96 0x00007fa5f0d8347d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
=======================
0x00007fa5da75c8c5 in opencog::AtomTable::getHandlesByType<std::back_insert_iterator<std::vector<opencog::Handle, std::allocator<opencog::Handle> > > >(std::back_insert_iterator<std::vector<opencog::Handle, std::allocator<opencog::Handle> > >, unsigned short, bool, bool) const [clone .local.32] (this=0x0, result=..., result@entry=..., type=type@entry=4, subclass=<optimized out>, 
    parent=parent@entry=true) at /home/mandeep/hr/opencog/atomspace/opencog/atomspace/AtomTable.h:220
220         if (parent && _environ)
````

misgeatgit, 2016-09-20:
@linas  please take a look at this issue as we need to demo some visualization of perception data from the robot.

linas, 2016-09-20:
OK, looking now. The last little bit is a null-pointer derference, and perhaps not entirely surprising: it is "well-known" that the rest API and python as well got mis-designed to use UUID's in a faulty manner. RIght now I am hoping that the fix will be obvious.

linas, 2016-09-21:
Hmm. none of the cogserver modules are even loaded, when I start the cogserver!

linas, 2016-09-21:
the config file issue fixed in opencog/opencog/pull/2444

linas, 2016-09-21:
How do you start the stack?  running `robot.sh --oc --cmt sophia` does not start the cogserver any more.

Running the cogserver plain doe not segfault.

linas, 2016-09-21:
Hmm. roscore is spweing error messages as fast as possible, for example:
```
  File "/home/ubuntu/hansonrobotics/HEAD/src/performances/scripts/wholeshow.py", li
ne 4, in <module>
    from transitions import *
ImportError: No module named transitions
```
There are also vast amounts of audio errors:
```
Traceback (most recent call last):                                      [1760/1886$
  File "/home/ubuntu/hansonrobotics/HEAD/src/sound/audio_stream/scripts/run.py", li
ne 29, in <module>
    with record_audio(channels, rate, chunk) as audio_stream:
  File "/usr/lib/python2.7/contextlib.py", line 17, in __enter__
    return self.gen.next()
```
Also blender is generating error messages:
```
Traceback (most recent call last):
  File "/home/ubuntu/hansonrobotics/sophia_blender_api/rigControl/blenderUI.py", line 78, in draw
    for i, emo in enumerate(bpy.evaAnimationManager.bones):
AttributeError: module 'bpy' has no attribute 'evaAnimationManager'
```

linas, 2016-09-21:
never mind. I did get the cogserver started. My bad.

AmeBel, 2016-09-21:
maybe you have to update https://github.com/opencog/ros-behavior-scripting/blob/master/scripts/opencog.conf , as per your recent changes in opencog repo?

linas, 2016-09-21:
To get a stack ttrace from gdb, one needs https://wiki.python.org/moin/DebuggingWithGdb
```
sudo apt-get install gdb python2.7-dbg
```
edit ~/.gdbint and add this
```
add-auto-load-safe-path  /usr/lib/debug/usr/bin/python2.7-gdb.py
```
Now we can look at the stack trace for real -- for  example, in stack frame 2 we can see this:
```
5682      __pyx_v_self->atomspace->get_handles_by_type(std::back_inserter(__pyx_v_handle_vector), __pyx_v_t, __pyx_v_subt);
(gdb) print __pyx_v_self->atomspace
$2 = (opencog::AtomSpace *) 0x0
```
not surprsing, we already knew that from up above, but its nice to see it printed in a pretty format

linas, 2016-09-21:
On a running server, saying this at the py shell segfaults:
```
print ATOMSPACE.uuid
```

linas, 2016-09-21:
On a running server, saying this at the py shell segfaults:
```
print len(ATOMSPACE)
```
segfaults and gdb shows this:
```
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffc27fc700 (LWP 22618)]
opencog::AtomTable::getSize (this=0x0)
    at /home/ubuntu/hansonrobotics/opencog/atomspace/opencog/atomspace/AtomTable.cc:739
739         return _size;
```

from which we conclude that the ATOMSPACE does not actually have the cogserver atomspace attached to it.  This is strongly suggestive of a very very old python bug ...

linas, 2016-09-21:
I believe that this is a work-around:
```
import opencog.cogserver
print opencog.cogserver.get_server_atomspace()
ATOMSPACE = opencog.atomspace.AtomSpace()
```

that will set the atomspace to the cogserver's atomspace.  Now the REST api doesn't crash.  However, something is still screwy, because REST claims that the atomspace is empty, when it clearly cannot be.

linas, 2016-09-21:
BTW, be sure to remove /usr/local/share/opencog/python/opencog -- the stuff there is now stale ... and leaving it there will mess things up, as comparesd to the new install locatin

linas, 2016-09-21:
ATTENTION: I mi-typed the above: the wwork-around should have said
```
import opencog.cogserver
print opencog.cogserver.get_server_atomspace()
ATOMSPACE = opencog.cogserver.get_server_atomspace()
import web.api.restapi
web.api.restapi.Start().run(None, ATOMSPACE)
```

once yu do tis, then the REST API works, reports all the atoms, and doesn't crash and etc.

My apologies, it was a late-night cut-n-paste error.

linas, 2016-09-21:
Anyway, with the above work-around, you are good to go, things should work.  I think I know exactly where the actual bug is, and will fix it tommorow AM when I am no longer making late-night errors.

misgeatgit, 2016-09-21:
Thanks @linas! it is working now.

linas, 2016-09-21:
OK, so opencog/atomspace/pull/888 should fix this bug, once and for all.

Please note that there were also changes to cog-utils mostly unrelated to this bug ...

AmeBel, 2016-09-22:
Yes the bug above seems to be fixed, but a segfault occurs when running the following. 
```bash
>> rlwrap telnet localhost 17020
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
opencog> py
Entering python shell; use ^D or a single . on a line by itself to exit.
py> import web.api.restapi
py> web.api.restapi.Start().run(None, ATOMSPACE)
py> .
Exiting the shell
opencog> .
Connection closed by foreign host.

[ user@machine:~ ]
>> curl 'http://localhost:5000/api/v1.1/atoms?type=ListLink&includeIncoming=true&includeOutgoing=true'

```
If the first curl doesn't cause the segfault running multiple instance of them will.

__The Segfault__

```
(gdb) bt
#0  __pyx_getprop_7opencog_9atomspace_4Atom_name(_object*, void*) [clone .13554] (o=0x7f6f9dc42050, x=0x0) at /home/amen/hansonrobotics/opencog/atomspace/build/opencog/cython/opencog/atomspace.cpp:13284
#1  0x00007f71c0ea5e81 in _PyObject_GenericGetAttrWithDict (obj=0x7f6f9dc42050, name=0x7f71c0118e70, dict=0x0) at ../Objects/object.c:1439
#2  0x00007f71c0ee9b11 in PyEval_EvalFrameEx (f=f@entry=0x7f7104628bf0, throwflag=throwflag@entry=0) at ../Python/ceval.c:2256
#3  0x00007f71c0eed54d in PyEval_EvalCodeEx (co=<optimized out>, globals=<optimized out>, locals=locals@entry=0x0, args=args@entry=0x7f6f9dec6748, argcount=3, kws=kws@entry=0x0, kwcount=kwcount@entry=0, 
    defs=defs@entry=0x0, defcount=defcount@entry=0, closure=0x0) at ../Python/ceval.c:3252
#4  0x00007f71c0f226d0 in function_call (func=0x7f71bbd5f230, arg=0x7f6f9dec6730, kw=0x0) at ../Objects/funcobject.c:526
#5  0x00007f71c0e8ed43 in PyObject_Call (func=func@entry=0x7f71bbd5f230, arg=arg@entry=0x7f6f9dec6730, kw=<optimized out>) at ../Objects/abstract.c:2529
#6  0x00007f71c0f07577 in PyEval_CallObjectWithKeywords (func=0x7f71bbd5f230, arg=0x7f6f9dec6730, kw=<optimized out>) at ../Python/ceval.c:3889
#7  0x00007f71c12f4fed in opencog::PythonEval::call_user_function (this=this@entry=0x1d19ea0, moduleFunction=..., arguments=...)
    at /home/amen/hansonrobotics/opencog/atomspace/opencog/cython/PythonEval.cc:812
#8  0x00007f71c12f756e in opencog::PythonEval::apply_tv (this=this@entry=0x1d19ea0, as=as@entry=0x11c3400, func=..., varargs=...)
    at /home/amen/hansonrobotics/opencog/atomspace/opencog/cython/PythonEval.cc:897
#9  0x00007f71c19788cd in opencog::EvaluationLink::do_evaluate (as=as@entry=0x11c3400, pn=..., cargs=...) at /home/amen/hansonrobotics/opencog/atomspace/opencog/atoms/execution/EvaluationLink.cc:604
#10 0x00007f71c196c60f in opencog::EvaluationLink::do_eval_scratch (as=as@entry=0x11c3400, evelnk=..., scratch=scratch@entry=0x11c3400, silent=silent@entry=false)
    at /home/amen/hansonrobotics/opencog/atomspace/opencog/atoms/execution/EvaluationLink.cc:260
#11 0x00007f71c196cf47 in opencog::EvaluationLink::do_eval_scratch (as=as@entry=0x11c3400, evelnk=..., scratch=scratch@entry=0x11c3400, silent=silent@entry=false)
    at /home/amen/hansonrobotics/opencog/atomspace/opencog/atoms/execution/EvaluationLink.cc:315
#12 0x00007f71c1978590 in opencog::EvaluationLink::do_evaluate (as=as@entry=0x11c3400, evelnk=..., silent=silent@entry=false)
    at /home/amen/hansonrobotics/opencog/atomspace/opencog/atoms/execution/EvaluationLink.cc:484
#13 0x00007f71c197898b in opencog::EvaluationLink::do_evaluate (as=as@entry=0x11c3400, pn=..., cargs=...) at /home/amen/hansonrobotics/opencog/atomspace/opencog/atoms/execution/EvaluationLink.cc:535
#14 0x00007f71c196c60f in opencog::EvaluationLink::do_eval_scratch (as=as@entry=0x11c3400, evelnk=..., scratch=scratch@entry=0x11c3400, silent=silent@entry=false)
    at /home/amen/hansonrobotics/opencog/atomspace/opencog/atoms/execution/EvaluationLink.cc:260
#15 0x00007f71c196cf47 in opencog::EvaluationLink::do_eval_scratch (as=as@entry=0x11c3400, evelnk=..., scratch=scratch@entry=0x11c3400, silent=silent@entry=false)
    at /home/amen/hansonrobotics/opencog/atomspace/opencog/atoms/execution/EvaluationLink.cc:315
#16 0x00007f71c196d62b in opencog::EvaluationLink::do_eval_scratch (as=as@entry=0x11c3400, evelnk=..., scratch=scratch@entry=0x11c3400, silent=silent@entry=false)
    at /home/amen/hansonrobotics/opencog/atomspace/opencog/atoms/execution/EvaluationLink.cc:337
---Type <return> to continue, or q <return> to quit---
#17 0x00007f71c196cf47 in opencog::EvaluationLink::do_eval_scratch (as=as@entry=0x11c3400, evelnk=..., scratch=scratch@entry=0x11c3400, silent=silent@entry=false)
    at /home/amen/hansonrobotics/opencog/atomspace/opencog/atoms/execution/EvaluationLink.cc:315
#18 0x00007f71c196d679 in opencog::EvaluationLink::do_eval_scratch (as=as@entry=0x11c3400, evelnk=..., scratch=scratch@entry=0x11c3400, silent=silent@entry=false)
    at /home/amen/hansonrobotics/opencog/atomspace/opencog/atoms/execution/EvaluationLink.cc:455
#19 0x00007f71c196d62b in opencog::EvaluationLink::do_eval_scratch (as=as@entry=0x11c3400, evelnk=..., scratch=scratch@entry=0x11c3400, silent=silent@entry=false)
    at /home/amen/hansonrobotics/opencog/atomspace/opencog/atoms/execution/EvaluationLink.cc:337
#20 0x00007f71c196cf47 in opencog::EvaluationLink::do_eval_scratch (as=as@entry=0x11c3400, evelnk=..., scratch=scratch@entry=0x11c3400, silent=silent@entry=false)
    at /home/amen/hansonrobotics/opencog/atomspace/opencog/atoms/execution/EvaluationLink.cc:315
#21 0x00007f71c196d62b in opencog::EvaluationLink::do_eval_scratch (as=as@entry=0x11c3400, evelnk=..., scratch=scratch@entry=0x11c3400, silent=silent@entry=false)
[... cut ...]
#63 0x00007f71ce5f3fe2 in GC_call_with_stack_base () from /usr/lib/x86_64-linux-gnu/libgc.so.1
#64 0x00007f71ceefe14c in ?? () from /usr/lib/libguile-2.0.so.22
#65 0x00007f71ce5f8d93 in GC_inner_start_routine () from /usr/lib/x86_64-linux-gnu/libgc.so.1
#66 0x00007f71ce5f3fe2 in GC_call_with_stack_base () from /usr/lib/x86_64-linux-gnu/libgc.so.1
#67 0x00007f71cec14184 in start_thread (arg=0x7f7105660700) at pthread_create.c:312
#68 0x00007f71ce94137d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111

```

linas, 2016-09-23:
Hmm. This should be a distinct bug report, but OK, investigating.

linas, 2016-09-23:
Very hard to reproduce, takes me 10-20 minutes of doing curl.  I get a very different stack trace, mine has no guile or EvaluationLink or even opencog::SchemePrimitive in it.  However, the end-point is very similar:
```
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff79a46700 (LWP 23747)]
__pyx_getprop_7opencog_9atomspace_4Atom_out(_object*, void*) [clone .13051] (
    o=0x7fff780b88e8, x=0x0)
    at /home/ubuntu/hansonrobotics/opencog/atomspace/build/opencog/cython/opencog/atomspace.cpp:13358
13358	  return __pyx_pw_7opencog_9atomspace_4Atom_3out_1__get__(o);
(gdb) bt
#0  __pyx_getprop_7opencog_9atomspace_4Atom_out(_object*, void*) [clone .13051] (
    o=0x7fff780b88e8, x=0x0)
    at /home/ubuntu/hansonrobotics/opencog/atomspace/build/opencog/cython/opencog/atomspace.cpp:13358
#1  0x00007fffea697e81 in _PyObject_GenericGetAttrWithDict (obj=0x7fff780b88e8, 
    name=0x7fffe95f6508, dict=0x0) at ../Objects/object.c:1439
#2  0x00007fffea62b09f in builtin_getattr.34122 (self=<optimized out>, 
    args=<optimized out>) at ../Python/bltinmodule.c:837
#3  0x00007fffea6de0d4 in call_function (oparg=<optimized out>, 
    pp_stack=0x7fff79a3f9d0) at ../Python/ceval.c:4020
#4  PyEval_EvalFrameEx (f=f@entry=0x7fff79cbea00, throwflag=throwflag@entry=0)
    at ../Python/ceval.c:2666
#5  0x00007fffea6de059 in fast_function (nk=<optimized out>, na=3, n=3, 
    pp_stack=0x7fff79a3fb50, func=0x7fff79cd3230) at ../Python/ceval.c:4106
#6  call_function (oparg=<optimized out>, pp_stack=0x7fff79a3fb50)
    at ../Python/ceval.c:4041
#7  PyEval_EvalFrameEx (f=f@entry=0x7fff79cbc420, throwflag=throwflag@entry=0)
    at ../Python/ceval.c:2666
```

This suggests a race in the cython bindings.

linas, 2016-09-23:
heh. And then there was this:
```
/build/buildd/gdb-7.7.1/gdb/value.c:3654: internal-error: value_fetch_lazy: Assertion `frame != NULL' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Quit this debugging session? (y or n) [answered Y; input not from terminal]
/build/buildd/gdb-7.7.1/gdb/value.c:3654: internal-error: value_fetch_lazy: Assertion `frame != NULL' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Create a core file of GDB? (y or n) [answered Y; input not from terminal]
Aborted (core dumped)
```

linas, 2016-09-24:
https://sourceware.org/bugzilla/show_bug.cgi?id=18074

linas, 2016-09-24:
a gdb marcro to print the python stack frame:
```
define pyf
  if $rip >= &PyEval_EvalFrameEx
   if $rip < &PyEval_EvalCodeEx
    x/s ((PyStringObject*)f->f_code->co_filename)->ob_sval
    x/s ((PyStringObject*)f->f_code->co_name)->ob_sval
    echo py line #
    p f->f_lineno
   end
 end
end
document pyf
show python stack frame
end
```

linas, 2016-09-24:
Some debugging notes:
```
#0  __pyx_getprop_7opencog_9atomspace_4Atom_out(_object*, void*) [clone
.13051] (
    o=0x7fff78b15cb0, x=0x0)
    at
/home/ubuntu/hansonrobotics/opencog/atomspace/build/opencog/cython/opencog/atomspace.cpp:13358
13358   return __pyx_pw_7opencog_9atomspace_4Atom_3out_1__get__(o);

static PyObject *__pyx_getprop_7opencog_9atomspace_4Atom_out(PyObject
*o, CYTHON_UNUSED void *x) {
  return __pyx_pw_7opencog_9atomspace_4Atom_3out_1__get__(o);
}

o is supposed to be:
(struct __pyx_obj_7opencog_9atomspace_Atom *)
struct __pyx_obj_7opencog_9atomspace_Atom {
  PyObject_HEAD
  opencog::Handle *handle;
  struct __pyx_obj_7opencog_9atomspace_AtomSpace *atomspace;
  PyObject *_atom_type;
  PyObject *_name;
  PyObject *_outgoing;
};

(gdb) p ((struct __pyx_obj_7opencog_9atomspace_Atom *)o)->_name
$33 = (PyObject *) 0x7fffeaaa02d0 <_Py_NoneStruct>
(gdb) p ((struct __pyx_obj_7opencog_9atomspace_Atom *)o)->_outgoing
$34 = (PyObject *) 0x7fffeaaa02d0 <_Py_NoneStruct>
(gdb) p ((struct __pyx_obj_7opencog_9atomspace_Atom *)o)->_atom_type
$35 = (PyObject *) 0x7fffeaaa02d0 <_Py_NoneStruct>

(gdb) p ((struct __pyx_obj_7opencog_9atomspace_Atom *)o)->handle->_ptr
$36 = {<std::__shared_ptr> = {_M_ptr = 0x0, _M_refcount = {
      _M_pi = 0x0}}, <No data fields>}
```
 So cython is holding a Handle with a null pointer in it, and it is trying to get it's outgoing set. So its a classi null-pointer dereference. How is it that python is holding a bad handle?

linas, 2016-09-24:
OK, as far as I can tell, opencog/atomspace/pull/891 mostly fixes this bug. There is still one more bug in the webapi itself, it crashes when it gets a SIGPIPE, which it will get if you control-C curl while its running.

linas, 2016-09-24:
Here's the actual stack trace
```
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/local/lib/python2.7/dist-packages/web/api/restapi.py", line 45, in invoke
    self.api.run(host=IP_ADDRESS, port=PORT)
  File "/usr/local/lib/python2.7/dist-packages/web/api/apimain.py", line 75, in run
    self.app.run(debug=False, host=host, port=port)
  File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 843, in run
    run_simple(host, port, self, **options)
  File "/usr/local/lib/python2.7/dist-packages/werkzeug/serving.py", line 694, in run_simple
    inner()
```

and the fix is to get the rest code to handle this exception

linas, 2016-09-24:
OK, so opencog/opencog/pull/2452 fixes that bug -- I just got done testing. -- its stayed up for 5 minutes

I hope this is the last of these bugs