opencog/ros-behavior-scripting #110: pull req #91 broke something.

linas, 2016-09-24:
After merging pull req #91, I get the following being printed, as fast as possible, when I run
```
guile -l btree-psi.scm
``` 
This:
```
(Eva expresses amused emotion for 3.0 seconds, with intensity 0.5 )
(Eva expresses amused emotion for 3.0 seconds, with intensity 0.5 )
(Eva expresses amused emotion for 3.0 seconds, with intensity 0.5 )
(Eva expresses amused emotion for 3.0 seconds, with intensity 0.5 )
(Eva expresses amused emotion for 3.0 seconds, with intensity 0.5 )
```
which clearly comes from this:
https://github.com/yenat/ros-behavior-scripting/blob/117954b7e4301223d6bf8c9997e99c0025d1364f/src/orchestrate.scm#L235-L249

I will try to disable this

linas, 2016-09-24:
bull req #111 stubs out the code until its fixed

AmeBel, 2016-09-26:
Saw into this, the rules should be ported to psi-dynamics rules, that will affect the level of positive/negative valence or some other cpm parameter. @eddiemonroe comments?

linas, 2016-11-05:
I think pull req #120 might fix this. It certainly removes the stubbed-out code.

linas, 2017-03-22:
closing I assume this is fixed.

closed 2017-03-22 by linas