opencog/ros-behavior-scripting #125: Unbound variable: request-eye-contact-state

linas, 2016-11-17:
I rebuilt everything from scratch today, and, on powering up, I get this crash:
```
2981: 9 [try-module-autoload (opencog eva-behavior) #f]
2320: 8 [save-module-excursion #<procedure 2019cf0 at ice-9/boot-9.scm:2982:17 ()>]
3001: 7 [#<procedure 2019cf0 at ice-9/boot-9.scm:2982:17 ()>]
In unknown file:
   ?: 6 [primitive-load-path "opencog/eva-behavior" ...]
In ice-9/boot-9.scm:
2320: 5 [save-module-excursion #<procedure 242fbc0 at ice-9/boot-9.scm:3961:3 ()>]
3970: 4 [#<procedure 242fbc0 at ice-9/boot-9.scm:3961:3 ()>]
1645: 3 [%start-stack load-stack #<procedure 21c4440 at ice-9/boot-9.scm:1666:2 ()>]
1650: 2 [#<procedure 24bc3f0 ()>]
In unknown file:
   ?: 1 [primitive-load-path "opencog/eva-behavior/behavior.scm"]
In opencog/eva-behavior/behavior.scm:
 136: 0 [#<procedure 2462960 ()>]

opencog/eva-behavior/behavior.scm:136:24: In procedure #<procedure 2462960 ()>:
opencog/eva-behavior/behavior.scm:136:24: In procedure module-lookup: Unbound variable: request-eye-contact-state
```

linas, 2016-11-18:
And another one too:
```
Traceback (most recent call last):
  File "./main.py", line 24, in <module>
    from face_track import FaceTrack
  File "/opencog/ros-behavior-scripting/face_track/face_track.py", line 28, in <module>
    from chatbot.msg import ChatMessage
ImportError: No module named chatbot.msg
```

AmeBel, 2016-11-18:
The `ImportError` is a sourcing `setup.bash` issue.

linas, 2016-11-18:
Yeah, I'll try to look at this later, its not urgent right now.

linas, 2017-03-22:
closing, this bug fixed itself. The chatbot message problem was fixed in pull req #167

closed 2017-03-22 by linas