opencog/ros-behavior-scripting #137: crash in ros-behavior-scripting

yantrabuddhi, 2016-12-13:
in time-map.scm, car wrong pair error 

stack trace :
[2016-12-13 10:55:18:996] [ERROR] Backtrace:
In ice-9/boot-9.scm:
 157: 16 [catch #t #<catch-closure 8ff79a0> ...]
In unknown file:
   ?: 15 [apply-smob/1 #<catch-closure 8ff79a0>]
   ?: 14 [opencog-extension cog-evaluate! ((DefinedPredicateNode "OpenPsi: loop$
)
)]
In ice-9/boot-9.scm:
 157: 13 [catch #t #<catch-closure 8f47f60> ...]
In unknown file:
   ?: 12 [apply-smob/1 #<catch-closure 8f47f60>]
In ice-9/boot-9.scm:
 157: 11 [catch #t #<catch-closure 8f47e00> ...]
In unknown file:
   ?: 10 [apply-smob/1 #<catch-closure 8f47e00>]
In opencog/openpsi/main.scm:
 126: 9 [psi-step]
In srfi/srfi-1.scm:
 576: 8 [map #<procedure 1774e00 at opencog/openpsi/main.scm:127:8 (d)> (# # # $
..)]
In ice-9/threads.scm:
  97: 7 [loop ((#))]
In opencog/openpsi/main.scm:
 104: 6 [act-and-evaluate #]
In unknown file:
   ?: 5 [opencog-extension cog-evaluate! (#)]
In ice-9/boot-9.scm:
 157: 4 [catch #t #<catch-closure 99de940> ...]
In unknown file:
   ?: 3 [apply-smob/1 #<catch-closure 99de940>]
In /home/mandeep/hr/opencog/ros-behavior-scripting/src/time-map.scm:
  56: 2 [look-turn-at-face (NumberNode "95213.000000")
 "gaze_at_face_point"]
In ice-9/boot-9.scm:
 102: 1 [#<procedure 8f9d740 at ice-9/boot-9.scm:97:6 (thrown-k . args)> wrong-$
ype-arg ...]
In unknown file:
   ?: 0 [apply-smob/1 #<catch-closure 99de900> wrong-type-arg ...]
ERROR: In procedure apply-smob/1:
ERROR: In procedure car: Wrong type argument in position 1 (expecting pair): ()
ABORT: wrong-type-arg

linas, 2016-12-13:
OK I think pull req #138 fixes this

linas, 2016-12-20:
Closing, I believe that this is now fixed.

closed 2016-12-20 by linas