opencog/ros-behavior-scripting #142: random exception

wenwei-dev, 2016-12-14:
ERROR: In procedure opencog-extension:
ERROR: Throw to key `C++-EXCEPTION' with args `("cog-evaluate!" "Backtrace:\nIn ice-9/boot-9.scm:\n 157: 16 [catch #t #<catch-closure 391bf80> ...]\nIn unknown file:\n   ?: 15 [apply-smob/1 #<catch-closure 391bf80>]\n   ?: 14 [opencog-extension cog-evaluate! ((DefinedPredicateNode \"OpenPsi: loop\")\n)]\nIn ice-9/boot-9.scm:\n 157: 13 [catch #t #<catch-closure 11facc0> ...]\nIn unknown file:\n   ?: 12 [apply-smob/1 #<catch-closure 11facc0>]\nIn ice-9/boot-9.scm:\n 157: 11 [catch #t #<catch-closure 11fab40> ...]\nIn unknown file:\n   ?: 10 [apply-smob/1 #<catch-closure 11fab40>]\nIn opencog/openpsi/main.scm:\n 126: 9 [psi-step]\nIn srfi/srfi-1.scm:\n 576: 8 [map #<procedure 11fa5a0 at opencog/openpsi/main.scm:127:8 (d)> (# # # ...)]\nIn opencog/openpsi/main.scm:\n 134: 7 [#<procedure 11fa5a0 at opencog/openpsi/main.scm:127:8 (d)> #]\nIn opencog/openpsi/action-selector.scm:\n 239: 6 [default-per-demand-action-selector #]\nIn unknown file:\n   ?: 5 [filter #<procedure 11b4100 at opencog/openpsi/rule.scm:387:8 (x)> #]\nIn opencog/openpsi/rule.scm:\n 347: 4 [is-satisfiable? #]\n 334: 3 [psi-satisfiable? #]\nIn unknown file:\n   ?: 2 [opencog-extension cog-evaluate! (#)]\nIn ice-9/boot-9.scm:\n 102: 1 [#<procedure 1e61d00 at ice-9/boot-9.scm:97:6 (thrown-k . args)> C++-EXCEPTION ...]\nIn unknown file:\n   ?: 0 [apply-smob/1 #<catch-closure 11fab00> C++-EXCEPTION ...]\n\nERROR: In procedure apply-smob/1:\nERROR: In procedure cog-evaluate!: Don't know how to do arithmetic with this: (SetLink\n) ; [9223565550901260237][0]\n (/home/hr/hansonrobotics/opencog/atomspace/opencog/atoms/execution/EvaluationLink.cc:86)\nABORT: C++-EXCEPTION\n (/home/hr/hansonrobotics/opencog/atomspace/opencog/guile/SchemeEval.cc:1226)")'.

linas, 2016-12-15:
This:
```
In opencog/openpsi/rule.scm:
 347: 4 [is-satisfiable? #]
 334: 3 [psi-satisfiable? #]
```
means that the problem is at line 334 of the file `rule.scm`  which reads:
```
    (define result (cog-evaluate!
        (SatisfactionLink (AndLink (psi-get-context rule)))))
```

linas, 2016-12-15:
Working backwards: The print statement `how to do arithmetic with` is from line 87 of `atoms/execution/EvaluationLink.cc`.  which is called by line 118 which is called by line 272 which is trying to handle a `GreaterThanLink`  The GreaterThanLink expects two NumberNodes, one of the NumberNodes must be coming from a search which resulted in an empty set.

The GreaterThanLink is in the psi-context of some rule.

linas, 2016-12-15:
Can you please reproduce with a new pull of the atomspace?  The current exception trace does not provide enough info to figure out which rule triggered this fault.  I just added some additional debugging info in https://github.com/opencog/atomspace/pull/1027  which should print out the rule that caused the issue.

yantrabuddhi, 2016-12-16:
ERROR: In procedure opencog-extension:
ERROR: Throw to key `C++-EXCEPTION' with args `("cog-evaluate!" "Backtrace:\nIn ice-9/boot-9.scm:\n 157: 16 [catch #t #<catch-closure 94ff200> ...]\nIn unknown file:\n   ?: 15 [apply-smob/1 #<catch-closure 94ff200>]\n   ?: 14 [opencog-extension cog-evaluate! ((DefinedPredicateNode \"OpenPsi: loop\")\n)]\nIn ice-9/boot-9.scm:\n 157: 13 [catch #t #<catch-closure 2f667a0> ...]\nIn unknown file:\n   ?: 12 [apply-smob/1 #<catch-closure 2f667a0>]\nIn ice-9/boot-9.scm:\n 157: 11 [catch #t #<catch-closure 2f664e0> ...]\nIn unknown file:\n   ?: 10 [apply-smob/1 #<catch-closure 2f664e0>]\nIn opencog/openpsi/main.scm:\n 126: 9 [psi-step]\nIn srfi/srfi-1.scm:\n 576: 8 [map #<procedure 2edd0e0 at opencog/openpsi/main.scm:127:8 (d)> (# # # ...)]\nIn opencog/openpsi/main.scm:\n 134: 7 [#<procedure 2edd0e0 at opencog/openpsi/main.scm:127:8 (d)> #]\nIn opencog/openpsi/action-selector.scm:\n 239: 6 [default-per-demand-action-selector #]\nIn unknown file:\n   ?: 5 [filter #<procedure 2ec7680 at opencog/openpsi/rule.scm:387:8 (x)> #]\nIn opencog/openpsi/rule.scm:\n 347: 4 [is-satisfiable? #]\n 334: 3 [psi-satisfiable? #]\nIn unknown file:\n   ?: 2 [opencog-extension cog-evaluate! (#)]\nIn ice-9/boot-9.scm:\n 102: 1 [#<procedure 2be1380 at ice-9/boot-9.scm:97:6 (thrown-k . args)> C++-EXCEPTION ...]\nIn unknown file:\n   ?: 0 [apply-smob/1 #<catch-closure 2f66480> C++-EXCEPTION ...]\n\nERROR: In procedure apply-smob/1:\nERROR: In procedure cog-evaluate!: Error while printing exception.\nABORT: C++-EXCEPTION\n (/home/mandeep/hr/opencog/atomspace/opencog/guile/SchemeEval.cc:1226)" "Function args:\n" ((DefinedPredicateNode "OpenPsi: loop")
))'.

yantrabuddhi, 2016-12-16:
[2016-12-16 06:41:21:350] [INFO] [OpenPsi] Finished evaluating of psi-rule (ImplicationLink (stv 1 1)
   (SequentialAndLink
      (SequentialAndLink
         (NotLink
            (DefinedPredicateNode "Skip Interaction?")
         )
         (NotLink
            (DefinedPredicateNode "tracking-salient?")
         )
         (DefinedPredicateNode "Someone visible?")
      )
      (DefinedPredicateNode "Interact with people")
   )
   (TrueLink
   )
)

[2016-12-16 06:41:21:371] [ERROR] Don't know how to do arithmetic with this: (SetLink
) ; [9223565550901260237][0]
 (/home/mandeep/hr/opencog/atomspace/opencog/atoms/execution/EvaluationLink.cc:86)
	Stack Trace:
	2: basic_string.h:293	  std::string::_M_data() const
	3: Logger.cc:493	  opencog::Logger::Error::operator()(char const*, ...)
	4: exceptions.cc:55	  opencog::StandardException::parse_error_message(char const*, __va_list_tag*, bool)
	5: exceptions.cc:81	  opencog::StandardException::parse_error_message(char const*, char const*, __va_list_tag*, bool)
	6: exceptions.cc:167	  opencog::SyntaxException::SyntaxException(char const*, char const*, ...)
	7: basic_string.h:539	  std::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()
	8: shared_ptr_base.h:545	__base_dtor ()
	9: EvaluationLink.cc:272	  opencog::EvaluationLink::do_eval_scratch(opencog::AtomSpace*, opencog::Handle const&, opencog::AtomSpace*, bool)
	10: EvaluationLink.cc:455	  opencog::EvaluationLink::do_eval_scratch(opencog::AtomSpace*, opencog::Handle const&, opencog::AtomSpace*, bool)
	11: EvaluationLink.cc:285	  opencog::EvaluationLink::do_eval_scratch(opencog::AtomSpace*, opencog::Handle const&, opencog::AtomSpace*, bool)
	12: shared_ptr_base.h:545	__base_dtor ()
	13: EvaluationLink.cc:485	  opencog::EvaluationLink::do_evaluate(opencog::AtomSpace*, opencog::Handle const&, bool)
	14: ExecSCM.cc:42	_ZL11ss_evaluatePN7opencog9AtomSpaceERKNS_6HandleE.92127()
	15: SchemeModule.cc:55	  opencog::FunctionWrap::as_wrapper_p_h(opencog::Handle)
	16: shared_ptr_base.h:545	__base_dtor ()
	17: SchemePrimitive.cc:163	  opencog::PrimitiveEnviron::do_call(scm_unused_struct*, scm_unused_struct*)
	18: ??:0	scm_vm_engine()
	19: ??:0	scm_call_1()
	20: ??:0	scm_filter()
	21: ??:0	scm_vm_engine()
	22: ??:0	scm_primitive_eval()
	23: ??:0	scm_eval()
	24: ??:0	scm_vm_engine()
	25: ??:0	scm_call_4()
	26: SchemeEval.cc:838	  opencog::SchemeEval::do_scm_eval(scm_unused_struct*, scm_unused_struct* (*)(void*))
	27: SchemeEval.cc:1183	  opencog::SchemeEval::do_apply_scm(std::string const&, opencog::Handle const&)
	28: SchemeEval.cc:1239	  opencog::SchemeEval::c_wrap_apply_tv(void*)
	29: ??:0	scm_at_abort()
	30: ??:0	scm_vm_engine()
	31: ??:0	scm_call_4()
	32: ??:0	scm_at_abort()
	33: ??:0	scm_c_with_continuation_barrier()
	34: ??:0	GC_call_with_stack_base()
	35: ??:0	scm_with_guile()
	36: SchemeEval.cc:1225	  opencog::SchemeEval::apply_tv(std::string const&, opencog::Handle)
	37: basic_string.h:539	  std::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()
	38: shared_ptr_base.h:545	__base_dtor ()
	39: EvaluationLink.cc:316	  opencog::EvaluationLink::do_eval_scratch(opencog::AtomSpace*, opencog::Handle const&, opencog::AtomSpace*, bool)
	40: shared_ptr_base.h:545	__base_dtor ()
	41: EvaluationLink.cc:455	  opencog::EvaluationLink::do_eval_scratch(opencog::AtomSpace*, opencog::Handle const&, opencog::AtomSpace*, bool)
	42: EvaluationLink.cc:485	  opencog::EvaluationLink::do_evaluate(opencog::AtomSpace*, opencog::Handle const&, bool)
	43: ExecSCM.cc:42	_ZL11ss_evaluatePN7opencog9AtomSpaceERKNS_6HandleE.92127()
	44: SchemeModule.cc:55	  opencog::FunctionWrap::as_wrapper_p_h(opencog::Handle)
	45: shared_ptr_base.h:545	__base_dtor ()
	46: SchemePrimitive.cc:163	  opencog::PrimitiveEnviron::do_call(scm_unused_struct*, scm_unused_struct*)
	47: ??:0	scm_vm_engine()
	48: ??:0	scm_pthread_mutex_lock()
	49: ??:0	scm_at_abort()

[... 45 lines cut ...]
	7: SchemeEval.cc:1226	  opencog::SchemeEval::apply_tv(std::string const&, opencog::Handle)
	8: basic_string.h:539	  std::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()
	9: shared_ptr_base.h:545	__base_dtor ()
	10: EvaluationLink.cc:316	  opencog::EvaluationLink::do_eval_scratch(opencog::AtomSpace*, opencog::Handle const&, opencog::AtomSpace*, bool)
	11: shared_ptr_base.h:545	__base_dtor ()
	12: EvaluationLink.cc:455	  opencog::EvaluationLink::do_eval_scratch(opencog::AtomSpace*, opencog::Handle const&, opencog::AtomSpace*, bool)
	13: EvaluationLink.cc:485	  opencog::EvaluationLink::do_evaluate(opencog::AtomSpace*, opencog::Handle const&, bool)
	14: ExecSCM.cc:42	_ZL11ss_evaluatePN7opencog9AtomSpaceERKNS_6HandleE.92127()
	15: SchemeModule.cc:55	  opencog::FunctionWrap::as_wrapper_p_h(opencog::Handle)
	16: shared_ptr_base.h:545	__base_dtor ()
	17: SchemePrimitive.cc:163	  opencog::PrimitiveEnviron::do_call(scm_unused_struct*, scm_unused_struct*)
	18: ??:0	scm_vm_engine()
	19: ??:0	scm_pthread_mutex_lock()
	20: ??:0	scm_at_abort()
	21: ??:0	scm_vm_engine()
	22: ??:0	scm_call_4()
	23: ??:0	scm_at_abort()
	24: ??:0	scm_c_with_continuation_barrier()
	25: ??:0	scm_current_processor_count()
	26: ??:0	GC_call_with_stack_base()
	27: ??:0	scm_all_threads()
	28: ??:0	GC_inner_start_routine()
	29: ??:0	GC_call_with_stack_base()
	30: pthread_create.c:312	start_thread()
	31: clone.S:113	clone()

yantrabuddhi, 2016-12-16:
guile: hashtab.c:227: weak_bucket_assoc: Assertion `(((scm_t_hashtable *) ((((scm_t_bits) (0? (*(SCM*)0=((((SCM *)((scm_t_cell *) (((scm_t_bits) (0? (*(SCM*)0=((((table))))): (((table)))))))) [((2))]))): (((SCM *)((scm_t_cell *) (((scm_t_bits) (0? (*(SCM*)0=((((table))))): (((table)))))))) [((2))]))))))->n_items) >= args.removed_items' failed.
/home/mandeep/hr/private_ws/scripts/btree.sh: line 20: 19965 Aborted                 (core dumped) $cmd

linas, 2016-12-18:
OK, I understand this problem now.   There is a race condition, where the StateLink can briefly not have a value during state update. As a result, the GetLink returns the empty set, and the GreaterThanLink pukes.   The solution is to have an atomic update of teh incoming set, so that the GetLink never fails.

linas, 2016-12-18:
Fix needs to be in atomspace, so will track this in issue https://github.com/opencog/atomspace/issues/1038

linas, 2017-01-02:
I beleive that this is now fixed by https://github.com/opencog/atomspace/pull/1053 -- please test.

linas, 2017-03-22:
closing I assume this is fixed

closed 2017-03-22 by linas