opencog/ros-behavior-scripting #151: Implement rules for combining saliency detection with face tracking

AmeBel, 2017-01-11:
- We want to distinguish “salient face events” from “salient non-face events" so, then, if there are no faces present, the behavior can be just to do saliency tracking and look around at salient events.

-  At the moment the system cannot recognize anything but faces — when we integrate object recognition software, then maybe it should have some tendency to look at objects it can recognize, as well as the tendency to look at salient stuff. But if there are faces present, then in general the face-tracking behavior should overrule the saliency tracking, unless something REALLY salient pops up.

- When we make the statement “Han, look at this!”  (or in general "ROBOTNAME, look at this!”) cause it to temporarily prioritize looking at salient events nearby the robot. And, “Look at that!” cause it to temporarily prioritize looking at salient events anywhere.

linas, 2017-01-11:
The code for "look at this" is mostly(?) done.  I briefly had some version of it working.

linas, 2017-01-11:
It's in the same file as "look at me" and "look at him", and the place where its unfinished is the part that tried to ground "me", "him" and "this" into the same kind of data structure.   For some of these, the data structure was a face ID, in other cases, the data structure was a 3D point.   So when I said "mostly" working", it works if you pick one or the other data structure, but since half the pipeline uses 3D points, and the other half uses face_id's, both styles could not work together at the same time.

I decided the best solution was to use object-id's everywhere -- i.e. convert face-id into an object-id, and then use object-ids for saliency. However, I only got party-way done in that conversion.

linas, 2017-01-11:
Also, "look at him" needed to be integrated with the manyears code -- I think I got part-way through with that, as well, so I could tell apart "me" (where sound is coming from) from "him" (a second visible face that isn't making sound).