opencog/ros-behavior-scripting #162: audio_stream.msg

weeksjr, 2017-03-09:
within the audio_power.py has the audio_stream.msg error been fixed?
or can someone point me to where it is defined? 

thanks

Jen

wenwei-dev, 2017-03-10:
I'm sure it's here https://github.com/hansonrobotics/HEAD/blob/master/src/sound/audio_stream/msg/audiodata.msg

linas, 2017-03-10:
Yeah, this is bad design .. this repo should not have any dependencies on HEAD, at all.

I'm not sure what a good solution is; a quick hack would be to copy the definition over this repo.  @weeksjr  do you know how to do this? would you care to do so?

linas, 2017-03-22:
Pull request #167 provides a work-around for this.

The correct solution is to split up HEAD  into a set of maintainable components; this is work in progress; until that split is made, we have no way of managing package dependencies.  After the split is made, we will be able to have a cmakefile that builds mandatory/optional components in an as-needed, as-available fashion.

AmeBel, 2017-08-08:
Resolved by multiple prs including https://github.com/opencog/ros-behavior-scripting/pull/175

closed 2017-08-08 by AmeBel